digidotcom/xbee-python #1: ImportError: cannot import name Serial

arunnediyasala, 2017-11-01:
While executing the program in Ubuntu 16.04 I am getting this error

Traceback (most recent call last):
  File "dgpython.py", line 1, in <module>
    from digi.xbee.devices import XBeeDevice
  File "build/bdist.linux-x86_64/egg/digi/xbee/devices.py", line 22, in <module>
  File "build/bdist.linux-x86_64/egg/digi/xbee/serial.py", line 15, in <module>
ImportError: cannot import name Serial

rubenmoral, 2017-11-02:
Hi @arunnediyasala,

This error means that Python cannot find the pyserial library, which is needed for the serial communication. In theory it should have been installed automatically when installing the XBee Python library, that's weird...

Anyway, you can install it manually:

`pip install pyserial`

Regards.

arunnediyasala, 2017-11-02:
Yes it is working now thanks

closed 2017-11-02 by arunnediyasala

arunnediyasala, 2017-11-02:
Thanks for you replay I have one more question regarding the program like
How to set up a co-coordinator using python

arunnediyasala, 2017-11-02:
Also can I able to get all the digi.xbee.devices details in python.

rubenmoral, 2017-11-03:
Glad it worked!

To set a module as coordinator, you should change the CE parameter to 1 (0 disabled, 1 enabled):

```python
from digi.xbee.devices import XBeeDevice
from digi.xbee.util import utils

device = XBeeDevice("COM1", 9600)
device.open()
device.set_parameter("CE", utils.int_to_bytes(1))
device.close()
```

In the API reference you can find all the available classes and methods of that module: http://xbplib.readthedocs.io/en/latest/api/digi.xbee.devices.html

arunnediyasala, 2017-11-03:
> devices.html

Thanks for your help

arunnediyasala, 2017-11-03:
> n/latest/api/digi.xbee.devices.html

My code

from digi.xbee.devices import XBeeDevice
from digi.xbee.util import utils
import serial
device = XBeeDevice("/dev/ttyUSB2", 9600)
device.open()
device.set_parameter("CE", utils.int_to_bytes(1))
device.close()

I am getting these exceptions

hon3 coordinator2.py
Traceback (most recent call last):
  File "coordinator2.py", line 6, in <module>
    device.set_parameter("CE", utils.int_to_bytes(1))
  File "/usr/local/lib/python3.5/dist-packages/digi/xbee/devices.py", line
1028, in dec_function
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/digi/xbee/devices.py", line
1353, in set_parameter
    raise ATCommandException("Error sending parameter, command status: " +
str(response.status))
digi.xbee.exception.ATCommandException: Error sending parameter, command
status: ATCommandStatus.INVALID_COMMAND

Please have a look

rubenmoral, 2017-11-03:
Hmmm, what XBee module do you have? That command status means that the CE parameter does not exist in the firmware of that module...

arunnediyasala, 2017-11-03:
XBee PRO 52B and the firmware I used as co -ordinator API

arunnediyasala, 2017-11-03:
Product family XBP24BZ7
firmware version 21A7 (Newest)

rubenmoral, 2017-11-03:
Oh, that makes sense. In those modules the CE parameter does not exist, you select the role (coordinator, router or end device) depending on the firmware you flash. Once done that, you don't need to do anything else, your module already is a coordinator.

You can follow the steps described at http://xbplib.readthedocs.io/en/latest/getting_started_with_xbee_python_library.html#zigbee-devices to form a ZigBee network.

arunnediyasala, 2017-11-03:
> en/latest/getting_started_with_xbee_python_library.html#zigbee-devices to

Hi
I can make my XBee as coordinator then I want to send data to other nodes
that are communicating to coordinator. So I am planning to execute
SendDataSample.py
<https://github.com/digidotcom/python-xbee/blob/master/examples/communication/SendDataSample/SendDataSample.py>
ReceiveDataSample
<https://github.com/digidotcom/python-xbee/tree/master/examples/communication/ReceiveDataSample>

I hope I can execute these program in XBee as a coordinator

arunnediyasala, 2017-11-03:
Hi,
I am using XBee Grove development board and I have 4 sensors ( 2 analog and
2 digital).2 analog sensors are working fine but digital sensor data I cam
not able to read I am using the following snippet

IOLINE_IN = IOLine.DIO2_AD2
IOLINE_IN0 = IOLine.DIO1_AD1
IO_LINE_IN = IOLine.DIO3_AD3
IO_LINE_OUT = IOLine.DIO4_AD4

device.set_io_configuration(IOLINE_IN, IOMode.ADC)
 device.set_io_configuration(IOLINE_IN0, IOMode.ADC)
 device.set_io_configuration(IO_LINE_IN, IOMode.DIGITAL_IN)
  device.set_io_configuration(IO_LINE_OUT, IOMode.DIGITAL_OUT_LOW)

for analog I am using GroveAD0/CB and Grove AD2
for digital I am using the grove Grove DIO4 & Grove DIO12 .

Question what is the address I have to use the Grove DIO4 & Grove DIO12  I
am used the following (IO_LINE_IN = IOLine.DIO3_AD3, IO_LINE_OUT =
IOLine.DIO4_AD4) .

Regards,
Arun Soman

rubenmoral, 2017-11-06:
Hey Arun,

You should use the following lines:

```python
IO_LINE_ADC_1 = IOLine.DIO0_AD0  # Connected to Grove AD0/CB
IO_LINE_ADC_2 = IOLine.DIO2_AD2  # Connected to Grove AD2
IO_LINE_IN = IOLine.DIO4_AD4  # Connected to Grove DIO4
IO_LINE_OUT = IOLine.DIO12  # Connected to Grove DIO12
```

Hope this helps!

arunnediyasala, 2017-11-06:
Thanks it works

arunnediyasala, 2017-11-06:
Hi Ruben,
Thanks for all your help I have one more question regarding the running of
the XBee. I want to run python program and read some sensor data through
XBee. Now I am using computer with a serial cable to connect  XBee. I want
a  device which can connect to XBee also it can  execute python programs to
do some task in XBee. I have some thing in mind like Raspberry Pi etc. But
I need some more cheaper one. Is there is in you mind please let me know.

Regards,
Arun Soman

arunnediyasala, 2017-11-07:
I have got a new exception
Transmit status: Broadcast source failed to hear a neighbor relay the
message
I got this exception from the coordinator  when coordinator is broadcasting
a message to the neighbor. How to solve this problem..

arunnediyasala, 2017-11-08:
Hi,
I want to catch this exception like
raise TimeoutException("Response not received in the configured timeout.")
digi.xbee.exception.TimeoutException

When I am executing my program my coordinator is exiting when there is no
zigBee are in range. I want to continue execution of my program with out
stop is there is any way to catch this exception.

Regards,
Arun Soman

diescalo, 2017-11-09:
You can catch the TimeoutException this way:

```python
from digi.xbee.exception import TimeoutException

# Initialize your device

try:
    # Execute code that may raise a TimeoutException here.
except TimeoutException as te:
    # Do whatever you want with the exception or just pass.
```

Regards

arunnediyasala, 2017-11-09:
I increased the timeout interval to 15 seconds so it is working fine now I
also use your try catch method to catch exception
local_device.set_sync_ops_timeout(15)

one more question How many nodes I can connect with a coordinator

hgonzaleDigi, 2017-11-10:
The response to that question is that it depends on the traffic you want the network to have, and also, the protocol you are using (in your case ZigBee).

In ZigBee, the default routing process works pretty well with networks of about 40 devices, but there are other routing methods used for larger networks, or even networks with different topologies. See some information here: https://www.digi.com/resources/documentation/DigiDocs/90002002/default.htm#Reference/r_zb_rf_packet_routing.htm
This documentation is for an advanced type user, and normally, the default routing process (Ad hoc On-demand Distance Vector (AODV) Mesh Routing) is more than enough to properly work in most environments.

Then, depending on whether you want your network to sleep or not, there are some guidelines to follow.
For this protocol, if the network is supposed to have sleeping nodes, there is a command called NC (https://www.digi.com/resources/documentation/DigiDocs/90002002/default.htm#Reference/r_cmd_NC.htm) to read the number of remaining end devices that you can connect to a given module.

This means that you should only connect around NC end devices to a coordinator or router. Although, if you don't need a sleeping network, you can probably connect more devices to a single coordinator/router.

Also, the network routing efficiency depends on the way your nodes will communicate with each other. Are they all going to send data to a central coordinator/aggregator? will the coordinator send messages to change the status of the end devices? There are many possibilities, and you can use a different aproach for each type of network.
If you want a big network, you should have XBees working in router mode between the coordinator and the end devices. You can connect several of those to a coordinator, and each of them can have multiple children devices. These devices can either be end devices, or other routers that can have more end devices or routers connected to them.

arunnediyasala, 2017-11-10:
> DigiDocs/90002002/default.htm#Reference/r_zb_rf_packet_routing.htm
> resources/documentation/DigiDocs/90002002/default.htm#
> Reference/r_cmd_NC.htm) to read the number of remaining end devices that

Hi,
I have 5 ZigBee I am running almost similar program like
RemoteADCSample.py but only one zigBee is responding. How can I communicate
with all ZingBees from one cordinator where I have to make change
remote_device = xbee_network.discover_device(REMOTE_NODE_ID)
if remote_device is None:
print("Could not find the remote device")
exit(1)

arunnediyasala, 2017-11-10:
I am using XBee SMT Grove want to connect I2C sensor on Grove DIO1/DIO11/I2C

The flowing I used for other application I want to know the corresponding
with I2C

IOLINE_IN0 = IOLine.DIO17

remote_device.set_io_configuration(IOLINE_IN0, IOMode.ADC)

arunnediyasala, 2017-11-13:
For communicating with many nodes from a coordinator should I create
coordinator node as DigiMesh device

local_device = DigiMeshDevice(PORT, BAUD_RATE)

arunnediyasala, 2017-11-13:
I am getting following error while setting local device as DigiMesh
local_device = DigiMeshDevice(PORT, BAUD_RATE)

Error
digi.xbee.exception.XBeeException: Error reading device information: Your
module seems to be XBeeProtocol.DIGI_MESH and NOT XBeeProtocol.DIGI_MESH.
Check if you are using the appropriate device class.

arunnediyasala, 2017-11-14:
I have a question about XBee SMT Grove Which grove I have to use for AD2
.In XBee THT I have Grove AD2 what is the corresponding grove in XBee SMT.

arunnediyasala, 2017-11-15:
Hi,
I am using XBee 2D SMT with firmware version 705B and hardware version is
3341 .I set CE as 1 but it is not detecting the neighboring nodes what I
have to do to get all other end device information

arunnediyasala, 2017-11-16:
Hi ,
I have 6 XBee nodes having sensors on it and a coordinator ,working fine
and if one of the node turned off the power then there is an exception is
generated  by the coordinator and it is failed to read data from other
nodes how to make sure that one node is alive or not , I am checking now
based on mac id but then also its coordinator fails when I am turned off
one of my XBee .
Coordinator is reading data values from XBee nodes using threading is there
is  any way to get to know any  node is failed so that I can avoid to
process that node in future.

rubenmoral, 2017-11-23:
Hi @arunnediyasala,

The GitHub issue tracker is not intended to provide general support about our products, if you still have some questions please contact directly to the Digi Support Team (tech.support@digi.com).

Regards.