digidotcom/xbee-python #5: async explicit data rx callback blocks

relihanl, 2017-11-27:
Hi...

I'm using the async explicit data received callback with Zigbee as follows:
`device.add_expl_data_received_callback(my_expl_data_received_callback)`

...and here is the content of the callback...
`def my_expl_data_received_callback(explicit_xbee_message):
    print("explicit data received")
    pprint(vars(explicit_xbee_message))

    print(" - Source endpoint:        %s"
          % utils.hex_to_string(utils.int_to_length(explicit_xbee_message.source_endpoint)))
    print(" - Destination endpoint:   %s"
          % utils.hex_to_string(utils.int_to_length(explicit_xbee_message.dest_endpoint)))
    print(" - Cluster ID:             %s"
          % utils.hex_to_string(utils.int_to_length(explicit_xbee_message.cluster_id)))
    print(" - Profile ID:             %s"
          % utils.hex_to_string(utils.int_to_length(explicit_xbee_message.profile_id)))
    print(" - Source address:         %s" % (explicit_xbee_message.remote_device.get_64bit_addr()))
    print(" - Data: %s", explicit_xbee_message.data.decode())
`

For some reason it blocks on the final statement when I try to read the data property. The following is the failure message:

`Traceback (most recent call last):
  File "digi-test.py", line 137, in <module>
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/serial/serialposix.py", line 472, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
OSError: [Errno 9] Bad file descriptor

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/digi/xbee/reader.py", line 297, in run
    raw_packet = self.__try_read_packet()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/digi/xbee/reader.py", line 642, in __try_read_packet
    xbee_packet[0] = self.__serial_port.read_byte()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/digi/xbee/serial.py", line 89, in read_byte
    byte = bytearray(self.read(1))
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/serial/serialposix.py", line 497, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: [Errno 9] Bad file descriptor

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/digi/xbee/reader.py", line 325, in run
    self.__xbee_device.serial_port.close()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/serial/serialposix.py", line 440, in close
    os.close(self.fd)
OSError: [Errno 9] Bad file descriptor`

I'm using v 1.0.0 of the library.

Liam

relihanl, 2017-11-27:
actually, scratch this. I removed the call to decode() and it seems to have unblocked this.

rubenmoral, 2017-11-28:
Hey @relihanl, can we close the issue then?

relihanl, 2017-11-28:
Certainly.sorry for the confusion

closed 2017-11-29 by rubenmoral