digidotcom/xbee-python #6: source and dest endpoint identical

relihanl, 2017-11-28:
in the explicit message object passed to the explicit data received callback, the "source_endpoint" and "dest_endpoint" are the same. It appears that the source endpoint is being copied from the dest_endpoint

i.e. in the following both printed values are identical

```python
def my_expl_data_received_callback(explicit_xbee_message):
         LOGGER.info("Source endpoint:        %s"
                 % utils.hex_to_string(utils.int_to_length(explicit_xbee_message.source_endpoint)))
         LOGGER.info("Destination endpoint:   %s"
                 % utils.hex_to_string(utils.int_to_length(explicit_xbee_message.dest_endpoint)))

```

relihanl, 2017-11-28:
Before anyone asks, I'm using a XB240ZB configured as a ZigBee Coordinator. F/W is 21A7

rubenmoral, 2017-11-28:
Hi @relihanl,

Thanks for reporting this problem, we have just fixed it in `master` (f420d8711b1fc305e1d5da7e62638a822ac14a62).

Best regards.

commit f420d871 (2017-11-28): fix #6: fix a bug when returning the source endpoint in the
ExplicitRXIndicatorPacket class
closed 2017-11-28 by rubenmoral