digidotcom/xbee-python #15: PacketListener does not handle length or frame-type fields being escaped

mikewadsten, 2018-01-08:
When using API mode 2 (escaped API mode), if either byte of the length field is escaped, then an incoming frame will either be misinterpreted (if the 4th byte happens to be a valid API frame type) or rejected.

Much like in #13, the issue lies in PacketListener.__try_read_packet:

```python
            packet_length = self.__serial_port.read_bytes(2)
            # Add packet length.
            xbee_packet += packet_length
            length = utils.length_to_int(packet_length)
```

The above code does not account for the fact that either byte (realistically, the second byte) of the length field might be escaped, in which case we will miscalculate the packet length.

In addition to this, the `factory.build_frame` function, and most if not all `create_packet` functions for different frame types, assumes that packet_bytearray begins with `0x7E <length_msb> <length_lsb> <frame_type>`, which is not necessarily true when in API escaped mode.

mikewadsten, 2018-01-08:
A generalized solution for this, #11, and #13, would be to have `__try_read_packet` un-escape all of the byte inside the frame, on the fly, before passing anything into `factory.build_frame`. This would also allow us to remove the `XBeeAPIPacket._unescape_data(raw)` calls from each API class.

closed 2018-01-18 by hgonzaleDigi

orenbell-nidec, 2019-05-24:
The documentation in create_packet (see below) explicitly states that it'll des-escape the byte array if needed, then fails to deliver on that promise. You shouldn't have to manually des-escape it yourself.

    If _OPERATING_MODE is API2 (API escaped) this method des-escape 'raw' and build the XBeePacket.
    Then, you can use :meth:`.XBeePacket.output` to get the escaped bytearray or not escaped.