digidotcom/xbee-python #22: digi.xbee.exception.InvalidOperatingModeException: Unsupported operating mode: AT mode

JBaczuk, 2018-01-22:
I'm trying to use this on a Macbook using Python3.  I have verified that the AP parameter is set to API Enabled [1] in XCTU.  When I try to run the following code:

```
from digi.xbee.devices import XBeeDevice

device = XBeeDevice('/dev/tty.usbserial-AL02DVXW', 115200)

print(device._operating_mode)
device.open()

while True:
    xbee_message = device.read_data()
    
    print(xbee_message)
```

I get `digi.xbee.exception.InvalidOperatingModeException: Unsupported operating mode: AT mode` even though it is in API Mode [1].  Any help is appreciated.

JBaczuk, 2018-01-22:
I think it may be because both devices must be in API mode, unfortunately I don't have access to the code on the second device, and when I put it in API mode, it appears to break the code and no longer transmits.

closed 2018-01-22 by JBaczuk

yxmagithub, 2019-09-26:
Jordan, do you solve this invalidOperatingModeException issue? if so How ?

JBaczuk, 2019-09-26:
@yxmagithub IIRC, I was not able to use this library for what I was trying to do. I think it only supports API mode, and I needed AT mode.