digidotcom/xbee-python #25: Configuration of PWM is not working

arunnediyasala, 2018-01-30:
Hi,
I tryto configure PWM  then I am getting the following error.
found_node.set_io_configuration(self.IO_LINE_PWM0, IOMode.PWM
File "/usr/local/lib/python3.5/dist-packages/digi/xbee/devices.py", line 624, in set_io_configuration
    self.set_parameter(io_line.at_command, bytearray([io_mode.value]))
  File "/usr/local/lib/python3.5/dist-packages/digi/xbee/devices.py", line 4835, in set_parameter
    raise ATCommandException("Error setting parameter, command status: " + response.status.description)
digi.xbee.exception.ATCommandException: Error setting parameter, command status: Invalid parameter

diescalo, 2018-01-31:
Hello @arunnediyasala . What XBee module are you using and what IO line are you trying to configure as PWM? Maybe that functionality is not supported in your module or in that specific IO line.

arunnediyasala, 2018-01-31:
Hi @diescalo I am using XBee S2D and I am trying to configure pin number 10
self.IO_LINE_PWM0 = IOLine.DIO10_PWM0
...............
......................
found_node.set_io_configuration(self.IO_LINE_PWM0, IOMode.PWM)
........................
............................
self.m[i].set_pwm_duty_cycle(self.IO_LINE_PWM0, 50)

diescalo, 2018-02-01:
I think that the module you are using only allows you to use the PWM for RSSI. The error you are getting ("Invalid parameter") means that the value provided in the command (2 in this case) is not supported.

closed 2018-07-11 by rubenmoral