digidotcom/xbee-python #27: Locla device serial port keeps closing

pdtwonotes, 2018-02-10:
On Ubuntu Linux with a Series-2 device in Coordinator mode.
`xbee = ZigBeeDevice('/dev/Zigbee', 9600 )
 xbee.open()
`
I do a few operations, which work fine, including
`xbee.send_data( remote, data )
`
But the next time I do `send_data` I get this exception:
`XBee device's serial port closed
`

rubenmoral, 2018-02-12:
Hi @pdtwonotes,

Could you please send us the complete stack trace of the exception? That would help us to understand why the port is being closed.

Thanks.

pdtwonotes, 2018-02-12:
I narrowed it down to class `PacketListener` line 327, where there is an `Exception` handler that picks up exceptions coming from the `packet_received` and `user_callback` calls.  So exceptions happening in _my_ code were being picked up here, and the handler closes the serial port.   Picking up the exception also prevented me even knowing my code was getting exceptions, so execution just continued on and eventually it would try to use the port again, resuling in the "port is closed" exceptions.

I fixed the problem by adding my own `logger` handler that recorded the _original_ exception details that get logged on line 328 so I could fix them.

The lessons here to users are:
- make sure your callbacks never throw an Exception back to the XBee API, unless you really want the port to close.

- Add a log handler so you mistakes do not get hidden.

rubenmoral, 2018-02-15:
Thanks for the clarification, @pdtwonotes. We will add some documentation to our API to prevent other users from having the same problem.

closed 2018-02-15 by rubenmoral