digidotcom/xbee-python #28: Device is not listed with encryption enabled

kennethlimcp, 2018-02-13:
I am using the xbee zigbee PRO dev kit and things are working great when `EE` is set to `Disabled [0]`. 

Changing `EE` to `Enabled [1]`, however, causes the remote nodes not to be listed in the `local_device.get_network()` process. Using `XCTU`, the nodes and master are able to list out all the devices though.

Is there a bug in the SDK?

hgonzaleDigi, 2018-02-13:
Hi @kennethlimcp,

We have looked into this issue and we were able to discover our remote devices once the encryption was enabled.

Please, verify that all your nodes have the Encryption enabled (EE = 1) and also, make sure that the encryption key is properly set on all devices (KY should be the same on all your modules).

In case you are only changing EE = 1 on your local device, then it makes sense that it cannot see the rest of the modules in the "network". Technically it will not be part of the same network since it is communicating using encryption and the rest of the nodes aren't.

Again, we have verified that once the encryption is enabled, both XCTU and this library behave the same way.

kennethlimcp, 2018-02-13:
I changed `EE` for both the XStick and end devices. Does `KY` need to be set manually? That is only required if `trust center` is selected or I might be wrong.

hgonzaleDigi, 2018-02-13:
Well, the best practices indicate that KY should be set manually on all the devices in the network.

This is how it works:
1. Encryption is enabled on all the devices (EE=1)
2. NK should be set on the coordinator. This value should be 0 (random key generated by the coordinator - most secure case)
3. KY should be set to the same value on all the devices (if left as default, which is 0, the process is not as secure). The use for KY is to encrypt the network key when joining the network so no security holes are created. In this case, if KY is 0, the coordinator will send the network key (NK) in the clear to each device that tries to join, whereas if KY is manually set, it will be used to encrypt the NK value so the network key will always remain secure in every network join process. You should consider doing this manually if you want security to be robust in your network and avoid anyone entering it without permission.
4. Finally, remember to write all the changes in non volatile memory (send WR command) so all these changes are permanent.

To answer your question regarding the trust center, when using zigbee network and APS encryption are supported regardless of whether you use a trust center or not. The follwing examples may help you: https://www.digi.com/resources/documentation/DigiDocs/90002002/default.htm#Tasks/t_security_examples.htm%3FTocPath%3DZigbee%2520security%7CImplement%2520security%2520on%2520the%2520XBee%252FXBee-PRO%2520Zigbee%2520RF%2520Module%7C_____6

The documentation there should help you understand how to work with the network in that manner.

If you keep having problems, please, let us know.

kennethlimcp, 2018-02-19:
@hgonzaleDigi it did not work for me:

```
Successfully built digi-xbee
Installing collected packages: pyserial, digi-xbee
Successfully installed digi-xbee-1.1.0 pyserial-3.4
(python-demo-ImaXBV93) bash-3.2$ python3 master.py 
 +---------------------------------------------+
 | XBee Python Library Discover Devices Sample |
 +---------------------------------------------+

Discovering remote XBee devices...
Device discovered: xxxxxxxxxxxxxx5B5D -  
Discovery process finished successfully.
```

The device listed is the coordinator.

From XCTU:

[image]

The coordinator is able to find the router via wireless.

Settings on coordinator:
- EE: 1
- EO: 0
- KY: 32 hexadecimal
- NK: 0

Settings on router:
- EE: 1
- EO: 0
- KY: 32 hexadecimal

kennethlimcp, 2018-02-22:
@hgonzaleDigi is there anyone I can speak to to have it resolved? We are looking to purchase 30 more nodes but this is the only blocker.

brandonmoser, 2018-02-22:
Kenneth,

Please send a request to Tech.Support@digi.com <mailto:Tech.Support@digi.com>. They will open a formal support case.

Thanks
Brandon

closed 2018-02-22 by rubenmoral

rubenmoral, 2018-02-22:
Hi @kennethlimcp,

We have fixed the issue (see commit c4a4c65fb144ed26130725b81604103a93339020), it will be included in the next release. In the meantime, you can download the source code from the `master` branch and install the library from there (see instructions at https://github.com/digidotcom/python-xbee/blob/master/README.rst#install-from-source).

Sorry for the inconvenience.

Best regards.

kennethlimcp, 2018-02-22:
> python-xbee/blob/master/README.rst#install-from-source).

Thanks for the updates. I'll test and report how it went. :)

ken,
sent on mobile

commit c4a4c65f (2018-02-22): fix #28: return an unknown XBee packet when the received packet is not
supported by the library instead of raising an exception