digidotcom/xbee-python #29: digi.xbee.exception.InvalidOperatingModeException: Unsupported operating mode: AT mode

cabas92, 2018-02-15:
I'm trying to use this library in a Raspberry Pi 3 with two modules XBEE S1. I have checked the correct operation of API mode [1] with a hyperterminal with different XBEE whit fixed PAN ID and CH. But, if I run the basic example:

```
from digi.xbee.devices import XBeeDevice
....
device = XBeeDevice('/dev/ttyS0', 9600)
device.open()
...
device.send_data_broadcast ("Hello XBees!")
...
device.close()

```
I get this error:
```
File "/home/pi/JairoC/ApiModeService.py", line 73, in <module>
      device.open()
File "usr/local/lib/python3.5/dist-packages/digi_xbee-1.1.0-py3.5.egg/digi/
xbee/devices.py", line 1262, in open
       raise InvalidOperatingModeException.from_operating_mode(self._operating_mode)
digi.xbee.exception.InvalidOperatingModeException: Unsupported operating mode: 
AT mode
```
I don't know what to do... Any help will be valued

ciel77, 2018-04-12:
I have similar issue with XBEE SX running on RPi 3. I am using 1 digi modem, and have another serial port device also connected via USB.

This AT mode exception happens every time I plug in both devices. But if only digi device is connected, it's working fine.

So is there a fix for multiple devices connected to USB ports?

ciel77, 2018-04-13:
I am using Digi XBee® SX RF Modems with latest version(1.1.0) on Raspberry Pi 3 B+ (Raspbian GNU/Linux 9 (stretch)). My python version is 3.5.3.

Somehow it times out when trying to send_packet_sync_and_get_response in when attempting do self.get_parameter("AP").

Spandyie, 2019-02-08:
Were you guys able to resolve this issue? @cabas92

cabas92, 2019-02-23:
Sorry, I haven't tried again for a solution. In my project, we updated the python library to the last version, configured the callback process for a received message and we used a try - except to send a message or start the XBEE module, and finally handling this exception (report error by the terminal or log file). Finally, I haven't review if the problem was solved, but my project goes fine.

ME-CostWatch, 2022-01-27:
I got this error when I plug the xbee into the socket but shifted up so the first two pins were not plugged in.