digidotcom/xbee-python #30: Join notification of ZigBee device

abbia18, 2018-02-16:
Hi,
if I start the discovery the Zigbee devices that are currently powered and connected to the network are correctly detected.
But if a device is "turned on" after the discovery process I don't receive the "join notification".
How can I capture this notification?
Thanks

hgonzaleDigi, 2018-02-16:
Hello @abbia18, 

According to the information you provided, it seems that the discovery process had stopped before the new device joined the network. Is this correct?

In that case, it is normal for the device not being detected. But, if the discovery process keeps running when the new device is powered on, it should be correctly discovered once it joins the network (it may take some time, though).

If your approach is not this one, and you just want to listen for the "join notification" packet (this means that the discovery process was finished), there are some things to take into account:

1. The new node that you add to the network should have the Join Notification parameter enabled (JN command should be 0x1). This way, when it joins to the network, it will send a Node Identification Indicator frame (0x95) to all the devices.
If that parameter is disabled, you won't be able to notice the joining event unless you are running a node discovery process.

2. Unfortunately, the Node Identification Indicator frame (0x95) is not supported natively by this library. The good news is that this behavior can be simulated just by adding a packet listener that waits for all the packets whose frame ID is 0x95. Here is an example of what that packet should look like so you can implement your listener and know the information to look for:
```
Node Identification Indicator (API 1)

7E 00 2A 95 00 13 A2 00 40 91 3B 75 18 38 02 18 38 00 13 A2 00 40 91 3B 75 52 54 2D 4B 59 4C 4F 2D 52 45 4E 00 FF FE 01 03 C1 05 10 1E 43

Start delimiter: 7E
Length: 00 2A (42)
Frame type: 95 (Node Identification Indicator)
64-bit source address: 00 13 A2 00 40 91 3B 75
16-bit source address: 18 38
Receive options: 02
16-bit remote address: 18 38
64-bit remote address: 00 13 A2 00 40 91 3B 75
NI string: 52 54 2D 4B 59 4C 4F 2D 52 45 4E (RT-KYLO-REN)
16-bit parent address: FF FE
Device type: 01 (Router)
Source event: 03
Digi profile ID: C1 05
Digi Manufacturer ID: 10 1E
Checksum: 43
```
Once you detect this packet, instantiate your remote device and add it to the list of devices of the network, thus simulating the behavior of the node discovery process.
This should be enough to detect the module joining to your network after being powered on.

Regards.

abbia18, 2018-02-16:
Thank you for you reply.
Since my approach is just to listen for the "join notification" packet, I will write a packet listener.

Unfortunatly I found another problem.
If I connect a new Zigbee device (for semplicity I call this device "ZigA") to an existing network and then I try to send a message to device "ZigA" the following exception happen "XBee device's serial port closed." 
Do you have any suggestions?

closed 2022-02-16 by tatianaleon