digidotcom/xbee-python #31: exception on device.close()

kennethlimcp, 2018-02-19:
```
Discovery process finished successfully.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/Users/xxxxxx/.local/share/virtualenvs/python-demo-ImaXBV93/lib/python3.6/site-packages/serial/serialposix.py", line 483, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
OSError: [Errno 9] Bad file descriptor

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/xxxxx/.local/share/virtualenvs/python-demo-ImaXBV93/lib/python3.6/site-packages/digi/xbee/reader.py", line 299, in run
    raw_packet = self.__try_read_packet(self.__xbee_device.operating_mode)
  File "/Users/xxxxx/.local/share/virtualenvs/python-demo-ImaXBV93/lib/python3.6/site-packages/digi/xbee/reader.py", line 671, in __try_read_packet
    xbee_packet[0] = self.__serial_port.read_byte()
  File "/Users/xxxxx/.local/share/virtualenvs/python-demo-ImaXBV93/lib/python3.6/site-packages/digi/xbee/serial.py", line 89, in read_byte
    byte = bytearray(self.read(1))
  File "/Users/xxxxx/.local/share/virtualenvs/python-demo-ImaXBV93/lib/python3.6/site-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: [Errno 9] Bad file descriptor

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/Cellar/python3/3.6.4_1/Frameworks/Python.framework/Versions/3.6/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/Users/xxxx/.local/share/virtualenvs/python-demo-ImaXBV93/lib/python3.6/site-packages/digi/xbee/reader.py", line 327, in run
    self.__xbee_device.serial_port.close()
  File "/Users/xxxxx/.local/share/virtualenvs/python-demo-ImaXBV93/lib/python3.6/site-packages/serial/serialposix.py", line 451, in close
    os.close(self.fd)
OSError: [Errno 9] Bad file descriptor

```

rubenmoral, 2018-02-22:
Hey @kennethlimcp,

I suspect this issue is related to #23. We have already fixed that, so please try again with the code from `master` and let us know if you still see this problem.

Best regards.

rubenmoral, 2018-07-11:
Closing this issue since the fix is already included in the latest release: https://github.com/digidotcom/python-xbee/releases.

To update the library through pip, run `pip install digi-xbee --upgrade`.

closed 2018-07-11 by rubenmoral