digidotcom/xbee-python #33: Cannot exit thread with device.close()

ciel77, 2018-03-15:
I started with `python-xbee/examples/communication/ReceiveDataPollingSample/ReceiveDataPollingSample.py` example by adding 
```

while True:
            xbee_message = device.read_data()
            if xbee_message is not None:
                print("From %s >> %s" % (xbee_message.remote_device.get_64bit_addr(),
                                         xbee_message.data.decode()))
                device.close()
                break
```
The code hang and I have to ctrl-c to exit

Then I got below message:
```
Exception ignored in: <module 'threading' from '/usr/lib/python3.5/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 1288, in _shutdown
    t.join()
  File "/usr/lib/python3.5/threading.py", line 1054, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.5/threading.py", line 1070, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
```
Any suggestion on how I can solve this?

FRVisser, 2018-04-04:
Same here, can never use device.close() without exception.

hgonzaleDigi, 2018-04-05:
Hello guys,

With the given information we have not been able to reproduce your problem, so to try it more accurately, you should give us the following information:

- Hardware and Firmware versions of your modules (HV and VR commands, respectively)
- The type of environment you are using. I mean, are you plugging your devices directly to a PC, or are you using another kind of product, like a Raspberry Pi?
- What code version are you running your example with? Is it the latest released version (Tag 1.1.0) or the latest code from master (Commit: 02febb6d0be7d0f48db252f2a124ff1f120e69d7)?
- We assume that you are running the exact example named above with the minor change of removing the try/finally blocks and including the close() and break actions in the if clause when a message is received, is that correct?

Regards,
Hector.

ciel77, 2018-04-12:
Thanks for reply Hector. 
1. I am using Digi XBee® SX RF Modems
2 & 3. I am running with latest version(1.1.0) on Raspberry Pi 3 B+ (Raspbian GNU/Linux 9 (stretch)). My python version is 3.5.3.
4. Correct

FRVisser, 2018-04-13:
Hi Hector,

For me it seems to be a duplicate of #23 and #24. It understand the is issue has been solved in master but not trough the pip repository. 

If I install the latest version on pip with 'pip install digi-xbee' I do get the same error message as the above. However, installing from git master branch solved the issue. 

1. I am using the xbee 868LP on the official digi development board
2. I am using MacOS with python3 in a virtualenv.
3. Versions:
osx 10.13.4 (17E199)

```
$ pip freeze
digi-xbee==1.1.0
pyserial==3.4
```

4. This is the complete code and complete output

```
$ python
Python 3.6.4 (default, Mar 22 2018, 13:54:22) 
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from digi.xbee.devices import DigiMeshDevice
>>> local_xbee = DigiMeshDevice("/dev/tty.usbserial-141430", 9600)
>>> local_xbee.open()
>>> local_xbee.close()
>>> Exception in thread Thread-1:
Traceback (most recent call last):
  File "/Users/fokkovisser/Google_Drive/Code/30MHz/Development/rangetest/digivirt/lib/python3.6/site-packages/serial/serialposix.py", line 483, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
OSError: [Errno 9] Bad file descriptor

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/fokkovisser/Google_Drive/Code/30MHz/Development/rangetest/digivirt/lib/python3.6/site-packages/digi/xbee/reader.py", line 299, in run
    raw_packet = self.__try_read_packet(self.__xbee_device.operating_mode)
  File "/Users/fokkovisser/Google_Drive/Code/30MHz/Development/rangetest/digivirt/lib/python3.6/site-packages/digi/xbee/reader.py", line 671, in __try_read_packet
    xbee_packet[0] = self.__serial_port.read_byte()
  File "/Users/fokkovisser/Google_Drive/Code/30MHz/Development/rangetest/digivirt/lib/python3.6/site-packages/digi/xbee/serial.py", line 89, in read_byte
    byte = bytearray(self.read(1))
  File "/Users/fokkovisser/Google_Drive/Code/30MHz/Development/rangetest/digivirt/lib/python3.6/site-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: [Errno 9] Bad file descriptor

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/Cellar/python/3.6.4_4/Frameworks/Python.framework/Versions/3.6/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/Users/fokkovisser/Google_Drive/Code/30MHz/Development/rangetest/digivirt/lib/python3.6/site-packages/digi/xbee/reader.py", line 327, in run
    self.__xbee_device.serial_port.close()
  File "/Users/fokkovisser/Google_Drive/Code/30MHz/Development/rangetest/digivirt/lib/python3.6/site-packages/serial/serialposix.py", line 451, in close
    os.close(self.fd)
OSError: [Errno 9] Bad file descriptor

```

hgonzaleDigi, 2018-04-13:
Hello all,

As @FRVisser said, this issue has been resolved in master, but a new release has not been generated. This means that PyPI does not have the latest packaged code to upgrade the library trhough pip.

We are working on generating a new version of the library to distribute through pip, and it will be released shortly.

In the meantime, if any of you needs the changes from master, you can update your python distributions manually. To do so:

1. Get the code from master.
2. In the root directory you will see a "digi" directory.
3. Go to your python distribution (<path_to_python3.x>/Lib/site-packages)
4. If the "digi" directory exists there from a previous distribution, uninstall it through pip, or remove it manually.
5. Add the "digi" directory just copied from master.

These steps should do it for you. Again, we are planning to do the official release soon, which would allow you to upgrade your library thrugh pip: pip install --upgrade digi-xbee

rubenmoral, 2018-07-11:
Closing this issue since the fix is already included in the latest release: https://github.com/digidotcom/python-xbee/releases.

To update the library through pip, run `pip install digi-xbee --upgrade`.