digidotcom/xbee-python #37: How to get RSSI of a message

h3ct0r, 2018-05-23:
Hi,

I would like to know how to get the RSSI between 2 nodes? I can see at the XCTU serial monitor the message with the RSSI, but I can't find how to get the RSSI with this python library.

Thanks!

h3ct0r, 2018-05-24:
Solved by using a different callback:

```
from digi.xbee.devices import XBeeDevice
from digi.xbee.packets.base import DictKeys

PORT = "/dev/ttyUSB1"
BAUD_RATE = 9600

def main():
    device = XBeeDevice(PORT, BAUD_RATE)

    try:
        device.open()

        def packet_received_callback(packet):
            packet_dict = packet.to_dict()
            api_data = packet_dict[DictKeys.FRAME_SPEC_DATA][DictKeys.API_DATA]
            data = api_data[DictKeys.RF_DATA]
            rssi = api_data[DictKeys.RSSI]
            address64 = api_data[DictKeys.X64BIT_ADDR]
            print("from: {}, RSSI: {}, Data: {}".format(address64, rssi, data.decode()))

        device.add_packet_received_callback(packet_received_callback)

        print("Waiting for data...\n")
        input()

    finally:
        if device is not None and device.is_open():
            device.close()

if __name__ == '__main__':
    main()
```

closed 2018-05-24 by h3ct0r

mchaudh4, 2019-03-08:
When i run the program, i get the ERROR. Please help me to find RSSI in the below program.
Waiting for data...

<DictKeys.RSSI: 'rssi'>
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.5/site-packages/digi/xbee/reader.py", line 325, in run
    self.__execute_user_callbacks(read_packet, remote)
  File "/home/pi/.local/lib/python3.5/site-packages/digi/xbee/reader.py", line 559, in __execute_user_callbacks
    self.__packet_received(xbee_packet)
  File "/home/pi/.local/lib/python3.5/site-packages/digi/xbee/reader.py", line 56, in __call__
    f(*args, **kwargs)
  File "StudyRSSICoordinator.py", line 19, in packet_received_callback
    rssi = api_data[DictKeys.RSSI]
KeyError: <DictKeys.RSSI: 'rssi'>

etherfi, 2019-08-02:
It fails because RSSI isn't present in the presented data. The author of this code neglected to mention that RSSI isn't always provided. It seems to only be available on WiFi modules, not the ZigBee ones.

shubham9436, 2020-05-27:
@h3ct0r did you manage to find the solution?

shubham9436, 2020-05-27:
@mchaudh4 did you find the solution?