digidotcom/xbee-python #39: Change _DEFAULT_TIMEOUT_SYNC_OPERATIONS Digimesh

FRVisser, 2018-06-07:
Hi All,

I am trying to properly change the default timeout for sending data with Digimesh as done in the example here: [SendDataSample.py](https://github.com/digidotcom/python-xbee/blob/8a9bf5febffc9f4418b1cc9cdc80a4b1082fad59/examples/communication/SendDataSample/SendDataSample.py#L45).

It sends the data fine, however when the device is new in the network or the address is not found it will always return a `digi.xbee.exception.TimeoutException`. But in the logs I see the Digimesh network replies its own error code, just a little later.

Moreover, if I alter the `_DEFAULT_TIMEOUT_SYNC_OPERATIONS ` in [devices.py](https://github.com/digidotcom/python-xbee/blob/8a9bf5febffc9f4418b1cc9cdc80a4b1082fad59/digi/xbee/devices.py#L56) up two seconds (6). I get a nice error report `digi.xbee.exception.XBeeException: Transmit status: Route not found.`

This is what I like to see. Unfortunately, there is no where to set the time out. Searching the code and the documentation didn't give any answers to this. 

Where can we set this timeout?

Much appreciated,
Fokko

rubenmoral, 2018-06-07:
Hi @FRVisser,

The library has a couple of methods to read and change the read timeout: `get_sync_ops_timeout()` and `set_sync_ops_timeout(timeout)`, so you can call them with an `XBeeDevice` object (or any subclass).

I hope this can help you.

Regards.

FRVisser, 2018-06-07:
Jeez. Once I know the function, I find the documentation. This fixed it.
Thanks!

closed 2018-06-07 by FRVisser