digidotcom/xbee-python #41: Response not received :Timeout exception

verossi, 2018-06-20:
Hi, 
I am getting a Timeout exception("Response not received in the configured timeout") error.
I am trying out the basic example of SendDataSample.py. I am using 2 Xbee pro Series 1 and the firmware version is 10ec. The python version is 3.6 and the xbee library version is 1.1.1.
These are the errors : 
......python-xbee/examples/communication/SendDataSample/SendDataSample.py", line 45, in main
    device.send_data(remote_device, DATA_TO_SEND).
......lib\site-packages\digi\xbee\devices.py", line 1544, in send_data
    return self._send_data_64(remote_xbee_device.get_64bit_addr(), data, transmit_options).
......\lib\site-packages\digi\xbee\devices.py", line 1028, in dec_function
    return func(self, *args, **kwargs).
.....lib\site-packages\digi\xbee\devices.py", line 1038, in dec_function
    response = func(*args, **kwargs).
....site-packages\digi\xbee\devices.py", line 1446, in _send_data_64
    return self.send_packet_sync_and_get_response(packet).
.....lib\site-packages\digi\xbee\devices.py", line 2423, in send_packet_sync_and_get_response
    raise TimeoutException("Response not received in the configured timeout.")
digi.xbee.exception.TimeoutException.

diescalo, 2018-06-20:
Hello @verossi 

The error could be caused because devices cannot communicate each other (maybe they are not configured with the same PAN ID or channel). Could you verify that they see each other please? You can do it using XCTU application.

If they can communicate each other, maybe the problem is that the timeout configured to get the answer is not enough (not likely, default timeout is 4 seconds I think). Anyway you can try changing it with:

`device.set_sync_ops_timeout(10)`

Regards!

verossi, 2018-06-21:
Thank you @diescalo for the reply. I got them working.  I reconfigured the devices to the parameter values that I was using before. 
Have another quick question. Can we configure the remote devices from the local devices using "set_parameter" as is shown for the Local XBee configuration example? 
I tried doing it but I suppose there should be a appropriate API where you would do a "send"(api) with the arguments as local device and destination device and the AT command to be sent(eg: ATAP etc) and the value to be set.
Is there an API as this?
Thank you!

diescalo, 2018-06-22:
You can configure remote devices the same way you do with local ones. All the configuration methods available for local devices are also compatible with remote ones. So for example you can do

```
local_device = XBeeDevice("COM1", 9600)
local_device .open()

local_device.set_parameter("NI",  bytearray("Naruto", 'utf8'))

xbee_network = local_device.get_network()
remote_device = xbee_network.discover_device("Remote")

remote_device.set_parameter("NI",  bytearray("Sasuke", 'utf8'))
```

The API underneath knows that the device you are configuring is a remote node and it executes the proper code, so you don't need to worry about that.

Regards!

diescalo, 2018-06-22:
@verossi Is this issue resolved for you? May I close your other issue too? I saw it was also related to a timeout problem.

Thanks!

verossi, 2018-06-24:
Hi @diescalo,
Thanks for the reply I have tried the set  parameters for remote device object but it does'nt work. 
I would share the code for that later, but I have another timeout exception and I am unable to understand as to why this is the case. I am currently trying out the IOSamplingSample.py example to read an ADC value by passing to it the analog signal from the RC circuit which in turn has the PWM pin as its input.  The code is as follows: 
`from digi.xbee.devices import XBeeDevice

from digi.xbee.io import IOLine, IOMode

# TODO: Replace with the serial port where your local module is connected to.
PORT = "COM5"
# TODO: Replace with the baud rate of your local module.
BAUD_RATE = 57600

REMOTE_NODE_ID = "REMOTE"
maxVoltage = 3.3
adcDecimalResolution = 1023
DIGITAL_LINE = IOLine.DIO3_AD3
ANALOG_LINE = IOLine.DIO2_AD2 #Pin number 18 on RemoteXBee
PWM_ONE=IOLine.DIO11_PWM1    #Pin number 7 on Remote XBee
IO_SAMPLING_RATE = 5  # 5 seconds.

def main():
    print(" +----------------------------------------------+")
    print(" | XBee Python Library Handle IO Samples Sample |")
    print(" +----------------------------------------------+\n")

    device = XBeeDevice(PORT, BAUD_RATE)

    device.open()

    # Obtain the remote XBee device from the XBee network.
    xbee_network = device.get_network()
    remote_device = xbee_network.discover_device(REMOTE_NODE_ID)
    print('connecting to remote device')
    if remote_device is None:
        print("Could not find the remote device")
        exit(1)
    print('connected to remote device')

    # Set the local device as destination address of the remote.
    remote_device.set_dest_address(device.get_64bit_addr())

    #remote_device.set_io_configuration(DIGITAL_LINE, IOMode.DIGITAL_IN)
    remote_device.set_io_configuration(ANALOG_LINE, IOMode.ADC)
    remote_device.set_io_configuration(PWM_ONE,IOMode.PWM)

    print('connected to remote device2')

    # Enable DIO change detection in the remote device.
    #remote_device.set_dio_change_detection({DIGITAL_LINE})

    # Register a listener to handle the samples received by the local device.
    coeff=maxVoltage/adcDecimalResolution
    ADCvalues = []
    print('input')
    dutyCycle= float(input())
    remote_device.set_pwm_duty_cycle(PWM_ONE,dutyCycle)
    device.set_sync_ops_timeout(5)

    while(True):
        #print('inside while.')

        #if (input()):
         #   dutyCycle= float(input())
        if dutyCycle not in range(0,101):
            if device is not None and device.is_open():
                device.close()
            thefile = open('test.txt', 'w')

            for item in ADCvalues:
                thefile.write("%s " % item)
                thefile.write("%s \n" % item)
            exit(1)

        # Enable periodic sampling every IO_SAMPLING_RATE seconds in the remote device.
        #remote_device.set_io_sampling_rate(IO_SAMPLING_RATE)

        ADC=remote_device.get_adc_value(ANALOG_LINE,)
        #print(ADC)
        AdcValue = coeff*ADC
        ADCvalues.append(dutyCycle)
        ADCvalues.append(AdcValue)
        print(AdcValue)

        #print(remote_device.get_adc_value(ANALOG_LINE))
main()`

Sorry for the messy code. I get an timeout exception for response not configured in the configured timeout, after a couple of values. I set the PWM once and try to get the AdcValue (which is back to volts). 

Anything that I'am doing wrong here? 
Thanks for the help in advance.

diescalo, 2018-06-25:
Hello @verossi ,

I have just tried your code with two 802.15.4 modules I have over here (I flashed them with firmware 10EC) and I got it working. Here is the output:

```
 +----------------------------------------------+
 | XBee Python Library Handle IO Samples Sample |
 +----------------------------------------------+

connecting to remote device
connected to remote device
connected to remote device2
input
100
0.3419354838709677
0.18064516129032257
0.14516129032258066
0.12580645161290321
0.11935483870967742
0.11290322580645161
0.11290322580645161
0.11935483870967742
0.12258064516129032
0.11935483870967742
0.11290322580645161
```
How far are your modules each other? For any reason they are losing communication between them. Have you tried increasing the 'sync operations timeout'?

verossi, 2018-06-26:
Hi @diescalo , The modules are very close (like on the same table). I also get the values but get the timeout exception for some reason. I have tried increasing the sync operations timeout but after around 20 readings(o/p values), there is a pause as if the local is waiting for the data to receive and then after the sync ops timeout period, the timeout exception. I have not configured any of them to sleep.
Thank you.

diescalo, 2018-06-26:
Hey @verossi,

What's the hardware version of the XBee modules? Could you send us the configuration profile of each module so we can replicate the setup here?

Thanks.

verossi, 2018-06-28:
Hi @diescalo , thank you for the reply. The hardware version for the 
Local XBee is : 1847 and  Remote XBee: 1847. The other parameters are as follows:

Parameters       Local                            Remote
CH                     C                                    C
ID                       1000                            1000
MY                     0                                   0
MM                   802.15.4+MAX... [0]        802.15.4+MAX.. [0] 
RR                      0                                     0
RN                     0                                       0
NT                     19                                       19
NO                      0                                   0
CE                     END DEVICE[0]                END DEVICE[0]
SC                     1FFE                                  1FFE
SD                      4                                       4
A1                      0000B                               0000B
A2                      000B                                  000B
AI                       0                                        0
EE                       DISABLE [0]                      DISABLE [0]
NI                      Local                                 REMOTE
PL                      HIGHEST[4]                       HIGHEST[4]
CA                    2C                                        2C
SM                   NO SLEEP[0]                        NO SLEEP[0]
ST                     1388                                     1388
SP                      0                                           0
DP                   3E8                                          3E8
SO                   0                                               0
BD                   57600                                      57600
NB                   NO PARITY[0]                        NO PARITY [0]
RO                    1                                             1
AP                   API ENABLED[1]                     API DISABLED[0]
D8                   DISABLED[0]                           DISABLED[0]
D7                   CTS FLOW CTRL[1]                 CTS FLOW CTRL[1]
D6                   DISABLED                                DISABLED
D5                   ASSOCIATED INDICATOR         ASSOCIATED INDICATOR
D4                   DISABLED                                 DISABLED
D3                    DISABLED                                 DISABLED
D2                    ADC                                           ADC
D1                    DISABLED                                  DISABLED
D0                    DISABLED                                 DISABLED
PR                      FF                                              FF
IU                      ENABLED                                 ENABLED
IT                     1                                                1
IC                    0                                                  0
IR                   3E8                                            3E8
P0                  PWM                                          PWM
P1                  PWM                                           PWM
PT                  FF                                                 FF
RP                  5                                                    28
IA                   FFFFFFFFFFFFFFFF                        FFFFFFFFFFFFFFFF
TO                  FF                                                  FF
T1                   FF                                                  FF
T2                   FF                                                  FF
T3                   FF                                                   FF
T4                   FF                                                   FF
T5                   FF                                                   FF
T6                   FF                                                   FF
T7                  FF                                                     FF
VR                  10EC                                               10EC
HV                1847                                                 1847
EC                 0                                                          0
EA                 783                                                   40
DD              10000                                                 10000
CT                64                                                       64
GT                 3E8                                                     3E8
CC                  2B                                                      2B

These are the configuration parameters.
Thank you.

verossi, 2018-07-02:
Hi @diescalo , any ideas how to solve this issue?

diescalo, 2018-07-02:
Hello @verossi 

Several advises about your setup & code:

- Why did you setup the IR value in the remote module if you are asking for the ADC value constantly? I mean, your `IR` configuration makes the local and remote modules to report an IO sample every second, but your code is already asking for IO samples to the remote module. If you don't need those automatic sample reports it I would configure it to 0.
- You are executing `ADC = remote_device.get_adc_value(ANALOG_LINE, )` constantly, which can be a bit stressing for the modules. I would wait some time (200ms or so) between requests.
- You can put `ADC = remote_device.get_adc_value(ANALOG_LINE, )` inside a try/except block. So if you get any timeout (or any other error) in one request, it can continue with the next one, without stopping the application:

You could rewrite the piece of code that reads the remote ADC value (the while loop) to something like this:
```
    while True:
        if dutyCycle not in range(0, 101):
            if device is not None and device.is_open():
                device.close()
            thefile = open('test.txt', 'w')

            for item in ADCvalues:
                thefile.write("%s " % item)
                thefile.write("%s \n" % item)
            exit(1)

        try:
            remote_adc = remote_device.get_adc_value(ANALOG_LINE)
            ADCvalue = coeff * remote_adc
            ADCvalues.append(dutyCycle)
            ADCvalues.append(ADCvalue)
            print(ADCvalue)
        except TimeoutException:
            print('Error getting remote ADC value > Timeout reading remote ADC')
        except Exception as e:
            print('Error getting remote ADC value > %s' % str(e))

        time.sleep(0.2)
```

Also, next time you can use XCTU to export a configuration profile instead of writing the configuration table manually =)  https://www.digi.com/resources/documentation/digidocs/90001458-13/default.htm#task/t_save_configuration_profile.htm%3FTocPath%3DConfigure%2520your%2520modules%7C_____6

Diego.

diescalo, 2018-07-11:
Hello @verossi,

Is your problem fixed with the suggestions I gave to you?

diescalo, 2018-07-26:
I'm closing this issue assuming the timeout problems you were having are fixed with the suggestions provided.

closed 2018-07-26 by diescalo