digidotcom/xbee-python #42: Unable to receive packets

kennethlimcp, 2018-06-26:
- Running python script on MacOS using Xstick2 ZB
- Using version `1.1.0` libraray
- Encryption is turned on

WFIW I also cannot discover any device though it works ok with XCTU.

[image]

[xbee_profile_backup.zip](https://github.com/digidotcom/python-xbee/files/2135271/xbee_profile_backup.zip)

kennethlimcp, 2018-06-26:
You see that `XCTU` receives the packets ok and a `SUCCESS` response is received by the node. 

However when switching over to the python script, it simply waits..

rubenmoral, 2018-06-26:
Hi @kennethlimcp,

I think the problem comes because you are also receiving `Route Record Indicator` packets, which are not supported by the library. In the version you are using (1.1.0), the library raises an exception when a not supported packet is received. We changed that behavior in https://github.com/digidotcom/python-xbee/commit/c4a4c65fb144ed26130725b81604103a93339020.

Please, use the latest release of the library (1.1.1) and let us know if you are still unable to receive the data packets.

Best regards.

DimitrisRigasPapakonstantinou, 2023-12-01:
Hello. I have the same problem. I installed the xbee libray using pip install. Should i do something different?