digidotcom/xbee-python #49: Timeout in IOSample callback

aronisch, 2018-07-13:
Hello !
I am trying to set a remote digital output each time an IOSample is received through a callback but I get a Timeout exception on set_io_configuration. Is the callback restricted ? I have a Digital IO Change Detection on DIO2 and I try to set DIO1 high or low.
I am using two Xbee3 RF micro in API1 mode, with the local one connected to a Raspberry Pi. The firmware version is 1003 with Zigbee protocol and here is my code.

```Python
from digi.xbee.devices import XBeeDevice, RemoteXBeeDevice
from digi.xbee.io import IOLine, IOMode
from digi.xbee.models.address import XBee64BitAddress

class SampleCallBackTest:

    def __init__(self):
        self.state = 0
        self.baseXbee = XBeeDevice("/dev/serial0", 115200)
        self.baseXbee.open()
        self.baseXbee.add_io_sample_received_callback(self.ioSampleCallback)
        self.remoteXbeeDev = RemoteXBeeDevice(self.baseXbee, XBee64BitAddress.from_hex_string(REMOTE_XBEE_ADDRESS))
        #self.baseXbee.set_sync_ops_timeout(5)

    def ioSampleCallback(self, ioSample, remoteXbee, sendTime):
        if self.state:
            remoteXbee.set_io_configuration(IOLine.DIO1_AD1, IOMode.DIGITAL_OUT_LOW)
            self.state = 0
        else:
            remoteXbee.set_io_configuration(IOLine.DIO1_AD1, IOMode.DIGITAL_OUT_HIGH)
            self.state = 1

sample = SampleCallBackTest()
```

rubenmoral, 2018-07-17:
Hi @aronisch,

Thanks for reporting this issue. We have fixed it (see commit e5a04ede1f9549a56ea7852553c20360edd37dce) and will be included in the next release. In the meantime, you can download the source code from the `master` branch and install the library from there (see instructions at https://github.com/digidotcom/python-xbee/blob/master/README.rst#install-from-source).

Sorry for the inconvenience.

Best regards.

closed 2018-07-17 by rubenmoral

aronisch, 2018-07-17:
Thanks for the update ! It works perfectly now

commit e5a04ede (2018-07-17): fix #49: execute all callbacks in parallel

The callbacks were being executed sequentially, which means that if a user
defined a callback and run any operation with the XBee device there, the
operation raised a TimeoutException.