digidotcom/xbee-python #57: Logger prints multiple times

FRVisser, 2018-08-16:
Hi,

I have a bit of strange issue. When I run my application the logger prints lines multiple times. For instance after while when I close my application:

```
/dev/tty.usbserial-1420 port closed
/dev/tty.usbserial-1420 port closed
/dev/tty.usbserial-1420 port closed
/dev/tty.usbserial-1420 port closed
/dev/tty.usbserial-1420 port closed
/dev/tty.usbserial-1420 port closed
/dev/tty.usbserial-1420 port closed
/dev/tty.usbserial-1420 port closed
/dev/tty.usbserial-1420 port closed
```

Even the last message I received was printed 7 times. As my program runs longer this number increases.

I checked, by adding some debug printing in the library, and the printed message is actually logged once but printed multiple times. And my program isn't leaking any memory or something. So everything does look fine.

I have been searching on it a little bit and the internet is full of the same questions regarding python logger. Like ([stackoverflow](https://stackoverflow.com/questions/47008864/how-to-prevent-logger-printing-same-information-multiple-times)):

> So every time you instantiate your class you add new handlers to the logger and every handler prints a line. 

Any idea how to solve this? Do you think this issue be related to having multiple logger handlers?

Thanks,
Fokko

FRVisser, 2018-08-16:
Ah and I checked the log. The log looks fine and doesn't have duplicated lines in it.

FRVisser, 2018-08-16:
Probing the `xbee.log.handlers` gives me the ever growing list:

`[<StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>]`

rubenmoral, 2018-09-07:
Hi @FRVisser,

We haven't had time to investigate your issue, we apologize for that.

Yes, there is a way. The Python's `logging` package allows you to change the logging level or even disable all logging calls. To do so, add these lines to your code:

```
import logging

[...]

logging.disable()
```

You can find more information at https://docs.python.org/3/library/logging.html#logging.disable.

Regards.

FRVisser, 2018-09-10:
Hi @rubenmoral,

Thanks for the suggestion! Unfortunately, your suggestion didn't have an effect. Any commands I send to logging seem not to help, this made my thinking where logger is set. 

We use logger in our code too, multiple times actually. Including a Flask web server, where logging was set to the Debug level. Changing this to a higher level, solved the issue. 

Now I  finally got some silence on the terminal :) Although it is unclear now, who is making al the logging handlers?

``` python
from digi.xbee.devices import DigiMeshDevice
xbee = DigiMeshDevice('/dev/tty.usbserial-DA011KM9',9600)
xbee.open()
xbee.log.handlers
```
outputs:

`[<StreamHandler <stderr> (NOTSET)>]`

Every time I send a packed a new one is added. Running the handlers list, exactly again will now display:

```
>>> xbee.log.handlers
[<StreamHandler <stderr> (NOTSET)>, <StreamHandler <stderr> (NOTSET)>]
```

So it definitely is the xbee library that adds the handlers

<br/><br/>
------------------
Returning to the type of messages I see in the terminal, I noticed a big difference as well. When I add the following nothing really happens.  

``` python
import logging
xbee.log.setLevel(logging.DEBUG)
```
It hasn't the same effect when I edit these settings the root logger to Debug. 

``` python
import logging
logger = logging.getLogger()
logger.setLevel(logging.DEBUG)
```
Now I see every Digi messages coming like these, very useful:
``` python
>>> /dev/tty.usbserial-DA011KM9RECEIVED    OperatingMode.API_MODE7E 00 1C 90 00 13 A2 00 41 5C D1 61 FF FE C2 64 61 74 61 64 61 74 61 64 61 74 61 64 61 74 61 C4
/dev/tty.usbserial-DA011KM9RECEIVED    DATA      0013A200415CD161  64 61 74 61 64 61 74 61 64 61 74 61 64 61 74 61
```

mlhales, 2019-02-06:
I just found this issue too. It turns out that it was already fixed in master at dad8aae0d744ae094e36f6f1987b0bf301d753b4 before you even asked your question. It hasn't been published as a new release though, so you will have to install from source, instead of using pip.

FRVisser, 2019-02-07:
🤦‍♂️ Well at least it is solved then!

aidanh010, 2019-02-27:
Tysm @mlhales this was making our application totally unusable. Here is all one has to do to take advantage of the newer code:

```bash
pip3 uninstall digi-xbee
pip3 install git+git://github.com/digidotcom/python-xbee#egg=digi-xbee
```

Digi should really push this onto PyPi, it would take about two seconds and remove a seriously frustration experience for new users.

tatianaleon, 2020-10-16:
Fix to this issue at c922ab1051aa6042ebde510f1564504fd14a3012 is already included in release 1.3.0.

closed 2020-10-16 by tatianaleon