digidotcom/xbee-python #59: Installing 1.1.1 error

arnulfojr, 2018-09-11:
```bash
$ pip install digi-xbee==1.1.1
Collecting digi-xbee==1.1.1
  Could not find a version that satisfies the requirement digi-xbee==1.1.1 (from versions: 1.0.0.1, 1.1.0.1, 1.1.1.1)
No matching distribution found for digi-xbee==1.1.1
```

However doing this works.
```bash
$ pip install digi-xbee==1.1.1.1
Collecting digi-xbee==1.1.1.1
  Using cached https://files.pythonhosted.org/packages/08/0c/2ed3b4af93dd957103d897c7899159b77d55b0809490669b25c5e469784b/digi-xbee-1.1.1.1.tar.gz
  Requested digi-xbee==1.1.1.1 from https://files.pythonhosted.org/packages/08/0c/2ed3b4af93dd957103d897c7899159b77d55b0809490669b25c5e469784b/digi-xbee-1.1.1.1.tar.gz#sha256=5bef6b1888886e99d1ec02ba35e1db17eb5f6f5b008e08cc85571f94df104acf, but installing version 1.1.1
Collecting pyserial>=3 (from digi-xbee==1.1.1.1)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Building wheels for collected packages: digi-xbee
  Running setup.py bdist_wheel for digi-xbee ... done
  Stored in directory: /home/asolis/.cache/pip/wheels/ee/8c/1f/a3490f75dd633035a32badb045abdb7d244009b919614859a1
Successfully built digi-xbee
Installing collected packages: pyserial, digi-xbee
Successfully installed digi-xbee-1.1.1 pyserial-3.4
```

When doing a `pip freeze` the version is `1.1.1`

knnleow, 2018-12-18:
i have the same problem. 
tried this  command failed.

    $ pip install digi-xbee==1.1.1
    Collecting digi-xbee==1.1.1
      Could not find a version that satisfies the requirement digi-xbee==1.1.1 (from versions: 1.0.0.1, 1.1.0.1, 1.1.1.1)
      No matching distribution found for digi-xbee==1.1.1

tried recommanded command. still fail

    $ pip install digi-xbee==1.1.1.1
    Collecting digi-xbee==1.1.1.1
    Using cached https://files.pythonhosted.org/packages/08/0c/2ed3b4af93dd957103d897c7899159b77d55b0809490669b25c5e469784b/digi-xbee-1.1.1.1.tar.gz
        Complete output from command python setup.py egg_info:
        Traceback (most recent call last):
          File "<string>", line 1, in <module>
        ModuleNotFoundError: No module named 'setuptools'    
        ----------------------------------------
    Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-bk14t2ad/digi-xbee/

my setup as follows...
    
    $ pip search digi-bee
       :
    digi-xbee (1.1.1)             - Digi XBee Python library
    
    $ pip --version
    pip 18.1 from /home/kuenn/.local/lib/python3.6/site-packages/pip (python 3.6)
    
    $ python3 --version
     Python 3.6.7

knnleow, 2018-12-18:
i am using ubuntu 18.04 on my laptop. version below: 

    Linux kuenn01 4.15.0-42-generic #45-Ubuntu SMP Thu Nov 15 19:32:57 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

knnleow, 2018-12-18:
forgive me for the noise... 
i solved by reinstall pip and working now

DigiEntmgmt, 2018-12-18:
> forgive me for the noise...
> i solved by reinstall pip and working now

So okay to close?

knnleow, 2018-12-19:
Yes

closed 2018-12-19 by DigiEntmgmt