digidotcom/xbee-python #79: Register Joining Device - 0x24 - support

AndreRenaud, 2019-07-23:
It doesn't look like there is support for the 0x24 packet type to enable registration of devices via their install code.
Is this supported by the library at the moment, or is it on the roadmap at all?
This is documented at:
https://www.digi.com/resources/documentation/digidocs/pdfs/90001539.pdf
(search for 'Example: Register a joining node using an install code')

From the code it looks like we'd need a new entry into https://github.com/digidotcom/xbee-python/blob/master/digi/xbee/packets/aft.py#L38, and probably a new packet class in https://github.com/digidotcom/xbee-python/blob/master/digi/xbee/packets/common.py

diescalo, 2019-08-16:
Hi @AndreRenaud, we've just made a new commit to give support for the **0x24 - Register Joining Device** and **0xA4 - Register Device Status** frames as well as wrapper methods to register and unregister devices in the **ZigBeeDevice** class. https://github.com/digidotcom/xbee-python/commit/e70cb7c81eefb19e993fd8fb33ee9ae8468ed0e1

There is not an official release with this feature, but the code has been added to the **master** branch of the repository, so you can download it to try and check how this new feature works.

Thanks!

tatianaleon, 2019-11-21:
These support was added in commit https://github.com/digidotcom/xbee-python/commit/e70cb7c81eefb19e993fd8fb33ee9ae8468ed0e1 and included in [release 1.3.0](https://github.com/digidotcom/xbee-python/releases/tag/1.3.0).

Closing this issue.

closed 2019-11-21 by tatianaleon