digidotcom/xbee-python #80: Log Output on Received Data

fmascarich, 2019-07-24:
The log output on data reception is quite annoying for integrating with other software, especially at high data transmission rates, would it be possible to add a flag at initialization to disable this debugging output?

FRVisser, 2019-10-15:
The big problem for me is that these messages are now logged as ERROR while should be INFO

FRVisser, 2019-10-15:
@fmascarich are you using callbacks to receive the data. The logging elements come [from there.](https://github.com/digidotcom/xbee-python/blob/68f238b665b913e20f6d7b69911fe12ee44b3990/digi/xbee/reader.py#L975) Although logging is disabled on my xbee device, my main python runtime logger is picking up these logging items. 

I commented out this line 975 to stop it.

FRVisser, 2019-10-15:
So it is actually logging for info `self._log.info`, but probably there is an error trying it.

tatianaleon, 2020-10-15:
Hi @fmascarich,

The log level for received frames has been changed from "info" to "debug" in `master` (commit 3ffaa0d39e07ff44b0b74948071b1edfcfe3f96d)

Anyway, you can completely disable the logger of the reader module:
```python
from digi.xbee.util import utils

utils.disable_logger("digi.xbee.reader")
```

Best Regards.

fmascarich, 2020-11-19:
@tatianaleon , thank you this solves this issue.

closed 2020-11-19 by fmascarich