digidotcom/xbee-python #97: Discovery not working

Aydin2, 2019-10-24:
Discovery example only prints the coordinator address. 
Coordinator is S2C, API 2
XCTU finds the endpoint device

securitybryan, 2020-06-09:
Could you post a code snippet so we can gain more insight into what you are doing?

sami-baroudi, 2020-07-02:
Hello,
I'm facing the same error:
I'm using a S2C in Coordinator mode + API2.
When launching the discovery, it returns an empty list. But XCTU detects the other device.

Code snippet :
[image]

securitybryan, 2020-07-03:
Have you tried bumping up the timeout value?   It is not uncommon for me to see 8-10 seconds for scan results.  I would set it to 25 for testing and then bump down from there.

tatianaleon, 2020-07-06:
@sami-baroudi, please see https://github.com/digidotcom/xbee-python/issues/184#issuecomment-653629609

closed 2020-10-15 by tatianaleon