digidotcom/xbee-python #103: Upgrade error

aleism2007, 2019-11-15:
hello, update to version 1.3 and now I can't run any examples

i have this error:

Traceback (most recent call last):
  File "DiscoverDevicesSample.py", line 70, in <module>
    main()
  File "DiscoverDevicesSample.py", line 34, in main
    device.open()
  File "D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\lib\site-packages\digi_xbee-1.3.0-py3.7.egg\digi\xbee\devices.py", line 2043, in open
  File "D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\lib\site-packages\digi_xbee-1.3.0-py3.7.egg\digi\xbee\devices.py", line 2065, in _do_open
  File "D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\lib\site-packages\digi_xbee-1.3.0-py3.7.egg\digi\xbee\devices.py", line 509, in read_device_info
  File "D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\lib\site-packages\digi_xbee-1.3.0-py3.7.egg\digi\xbee\devices.py", line 528, in _determine_role
  File "D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\lib\site-packages\digi_xbee-1.3.0-py3.7.egg\digi\xbee\devices.py", line 1584, in dec_function
  File "D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\lib\site-packages\digi_xbee-1.3.0-py3.7.egg\digi\xbee\devices.py", line 2121, in get_parameter
  File "D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\lib\site-packages\digi_xbee-1.3.0-py3.7.egg\digi\xbee\devices.py", line 225, in get_parameter
  File "D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\lib\site-packages\digi_xbee-1.3.0-py3.7.egg\digi\xbee\devices.py", line 315, in __send_parameter
  File "D:\Users\Alejandro\AppData\Local\Programs\Python\Python37-32\lib\site-packages\digi_xbee-1.3.0-py3.7.egg\digi\xbee\devices.py", line 334, in _check_at_cmd_response_is_valid
digi.xbee.exception.ATCommandException: There was a problem sending the AT command packet.

tatianaleon, 2019-11-15:
Could you tell us which XBee are you using? Firmware and hardware version?

aleism2007, 2019-11-15:
Hello, 
          the problem was that the XB3-24 had the firmware version 3001. I updated it to the version 3004 and the problem was solved.

Thanks

rasaffie, 2019-11-19:
Same error for me using XBee S2C with firmware 9002. Had to downgrade to version `1.2.0` to solve problem.

tatianaleon, 2019-11-19:
This is a bug, the problem is the synchronous sleep is not supported in DigiMesh S2C modules nor in XBee 3 3001 firmware version. This means the `SS` command does not exists for these firmwares, so the XBee initialization process fails when trying to determine the role.

I will open a PR to fix this in master. It will be released in the next version of the library.

tatianaleon, 2019-11-20:
Hi @aleism2007 and @rasaffie,

Thanks for reporting this issue. We fixed it in `master` (see commit https://github.com/digidotcom/xbee-python/commit/a10f3dfc7eb283f93c46ed3e559b9429825e5a51) and will be included in the next release.

In the meantime, you can download the source code from the master branch and install the library from there (see instructions at https://github.com/digidotcom/python-xbee/blob/master/README.rst#install-from-source).

Sorry for the inconvenience.

Best regards.

closed 2019-11-20 by tatianaleon