digidotcom/xbee-python #106: digi-xbee on raspberri pi 3

aang7, 2019-11-29:
Hello, I'm just trying to connect to my XBee module with a raspberry pi 3 and I can't. Some one knows how to achieve this?
I've already have installed the library and using python 3.7.3 within raspbian buster and nothing...

_Originally posted by @aang7 in https://github.com/digidotcom/xbee-python/issues/75#issuecomment-559623396_

tatianaleon, 2019-12-02:
Hi @aang7 ,

This might be related with issue https://github.com/digidotcom/xbee-python/issues/103.
You can try with the source code in master.

Sorry for the inconvenience.

tatianaleon, 2019-12-03:
Hi @aang7,

Thanks for reporting this issue. It is already fixed in master (see commit https://github.com/digidotcom/xbee-python/commit/a10f3dfc7eb283f93c46ed3e559b9429825e5a51) and will be included in the next release.

In the meantime, you can download the source code from the master branch and install the library from there (see instructions at https://github.com/digidotcom/python-xbee/blob/master/README.rst#install-from-source).

Sorry for the inconvenience.

Best regards.

closed 2019-12-03 by tatianaleon