digidotcom/xbee-python #109: device.open() returns raise TimeoutException(message="Response not received in the configured timeout.") on UART

UTSAnonymous, 2019-12-12:
Hi guys, 

I am encountering this issue where using the XBee module (XBee-Pro S2C 2.4GHz RF) with UART on the Jetson Nano returns this error
      -> raise TimeoutException(message="Response not received in the configured timeout.")
from the function _send_packet_sync_and_get_response. 

I tried to debug this by setting the XBee module to be on a known Digimesh network we are using on another setup and configure it to be on transparent mode. Using the UART serial port on the Jetson Nano we were able to receive data from that network. We also tried to use a USB-FTDI explorer board instead of a UART one and it works fine as well. The error only occurs when we are using UART. 

This issue seems to be similar to #7 however we are getting the same error when using UART only. 

Can only help me out with this issue?

`Traceback (most recent call last):
  File "xbee_library_test.py", line 37, in <module>
    main()
  File "xbee_library_test.py", line 28, in main
    xbee.open()
  File "/usr/local/lib/python3.6/dist-packages/digi/xbee/devices.py", line 2043, in open
    self._do_open()
  File "/usr/local/lib/python3.6/dist-packages/digi/xbee/devices.py", line 2065, in _do_open
    self.read_device_info()
  File "/usr/local/lib/python3.6/dist-packages/digi/xbee/devices.py", line 479, in read_device_info
    self._firmware_version = self.get_parameter(ATStringCommand.VR.command)
  File "/usr/local/lib/python3.6/dist-packages/digi/xbee/devices.py", line 1584, in dec_function
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/digi/xbee/devices.py", line 2121, in get_parameter
    return super().get_parameter(param, parameter_value=parameter_value)
  File "/usr/local/lib/python3.6/dist-packages/digi/xbee/devices.py", line 225, in get_parameter
    value = self.__send_parameter(parameter, parameter_value=parameter_value)
  File "/usr/local/lib/python3.6/dist-packages/digi/xbee/devices.py", line 313, in __send_parameter
    response = self._send_at_command(at_command)
  File "/usr/local/lib/python3.6/dist-packages/digi/xbee/devices.py", line 381, in _send_at_command
    answer_packet = self._send_packet_sync_and_get_response(packet)
  File "/usr/local/lib/python3.6/dist-packages/digi/xbee/devices.py", line 1782, in _send_packet_sync_and_get_response
    raise TimeoutException(message="Response not received in the configured timeout.")
digi.xbee.exception.TimeoutException: Response not received in the configured timeout.
`

tmrose, 2020-06-03:
Did you happen to have any luck solving this problem? I'm having the same issue with 9XTend-PKG-U.

UTSAnonymous, 2020-06-05:
@tmrose No actually, I wasn't quite sure what was going on as the same setup on the Raspberry Pi 3 works. I was stuck with USB at the movement but I'm trying to see if I get better luck using SPI

mroavi, 2020-10-07:
I was having the same issue. My problem was that the receiver module had the API mode disabled. Solved it by enabling it.

Artyrm, 2021-08-20:
.open() happens way before network communication, and deals solely with the serial port. Try maybe to communicate with your device with pure serial?

For example:
`minicom -D /dev/ttyAMA0 -b 9600
minicom -D /dev/ttyS0 -b 9600`

On typing "+++" device should respond with "OK".