digidotcom/xbee-python #122: Question about link quality

aminrd, 2020-01-30:
Hi there, 

I have two Xbee modules in an Xbee mesh netwwork that are communicating with each other. I want to show the link quality on my web application over time. I have installed the Betta version, in which, LinkQuality and Connection classes are added to `device.py`. 

Currently, when I call `digi.xbee.devices.Connection(device1, device2).lq_a2b`  it always returns `UNKNOWN_VALUE = -9999`. 

However, in XCTU application, I can see the link quality between two xbee modules using Range test. How can I somehow show connection quality (in dbm or % or ...) using python packages. 

Thank you in advance.

tatianaleon, 2020-05-14:
Hi @aminrd,

The class `Connection` is not intended to be instantiated in such way. The connection information is filled once the network is deeply discovered at least once. When this process finishes, you can get the connection by using `get_connections()` of your XBee network:

```python

    # Callback for discovered devices.
    def callback_device_discovered(remote):
        print("Device discovered: %s" % remote)

    # Callback for discovery finished.
    def callback_discovery_finished(status):
        if status == NetworkDiscoveryStatus.SUCCESS:
            print("Discovery process finished successfully.")
        else:
            print("There was an error discovering devices: %s" % status.description)

    [...]

    xbee = ZigBeeDevice(PORT, BAUD_RATE)
    xbee.open()

    xb_net = device.get_network()

    xb_net.add_device_discovered_callback(callback_device_discovered)
    xb_net.add_discovery_process_finished_callback(callback_discovery_finished)

    xb_net.start_discovery_process(deep=True)

    [...]

    if xb_net.has_devices():
        print("%s" % '\n    '.join(map(str, xb_net.get_connections())))

    [...]
```

closed 2020-05-26 by tatianaleon

shubham9436, 2020-05-28:
Hi @tatianaleon ,

Did you find how to determine the link quality between two devices?
```
connection = Connection(device, xb_net.get_connections())
print(connection.lq_b2a)
print(connection.lq_a2b)
```

I get `?` as output.

daescalona, 2020-05-28:
Hi @shubham9436 

As Tatiana said, you cannot directly instantiate or create a "Connection" object. The link quality between two nodes is only calculated/determined when performing a "deep" scan of the network. You can see an example on how to do a deep scan in the previous answer. 

Once the deep scan is complete, the network stores "Connection" objects with link quality information between nodes. You can extract the connection list from the network invoking the "get_connections()" method. That is the only way to determine the link quality of nodes using the API.

seanhodgson, 2021-06-07:
[test2.zip](https://github.com/digidotcom/xbee-python/files/6612334/test2.zip)
Hi I'm trying to get link quality from my xbee devices.  I have attached the code I'm using to test this.  I have used the directions @daescalona mentioned about configuring deep discovery and I still only get the following results:

Discovery process finished successfully.
{0013A20041B4954A - X3 >>> 0013A20041C7DAC2 - X1 [Active / Active]: ? / ?}
    {0013A20041B4954A - X3 >>> 0013A20041B498FA - X2 [Active / Active]: ? / ?}

Do you know what I'm doing wrong for this?  I can see my two nodes, but don't get link quality.

tatianaleon, 2021-06-08:
Hi @seanhodgson,

If you follow the [example above](https://github.com/digidotcom/xbee-python/issues/122#issuecomment-628733843), you must use protocol classes (`ZigBeeDevice` or `DigiMeshDevice`). Connections LQI or RSSI is only available for Zigbee and DigiMesh networks.

Try to change:
```python
device = XBeeDevice(PORT,BAUD_RATE)
```
To use:
* Zigbee modules:
```python
device = ZigBeeDevice(PORT,BAUD_RATE)
```
* DigiMesh modules:
```python
device = DigiMeshDevice(PORT,BAUD_RATE)
```

seanhodgson, 2021-06-08:
Thanks for the clarification.  I have replaced the XBeeDevice with DigiMeshDevice.  It opens the device, but at about the point it should discover one of the other DIGI xbee RF Modems I'm working with it throws the following exceptions:
`exception calling callback for <Future at 0x226f9605ac0 state=finished raised AttributeError>
Traceback (most recent call last):
  File "C:\Users\SeanHodgson\AppData\Local\Programs\Python\Python38\lib\concurrent\futures\_base.py", line 328, in _invoke_callbacks
    callback(self)
  File "C:\Users\SeanHodgson\AppData\Local\Programs\Python\Python38\lib\site-packages\digi\xbee\reader.py", line 91, in __execution_finished
    raise future.exception()
  File "C:\Users\SeanHodgson\AppData\Local\Programs\Python\Python38\lib\concurrent\futures\thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "C:\Users\SeanHodgson\AppData\Local\Programs\Python\Python38\lib\site-packages\digi\xbee\models\zdo.py", line 1773, in __fn_packet_cb
    self.__parse_data(frame.command_value)
  File "C:\Users\SeanHodgson\AppData\Local\Programs\Python\Python38\lib\site-packages\digi\xbee\models\zdo.py", line 1731, in __parse_data
    self._logger.debug("Neighbor of '%s': %s (relation: %s, rssi: -%s)", self._xbee,
AttributeError: 'NeighborFinder' object has no attribute '_xbee'`

Is there a better example I should be using for a starting point?

seanhodgson, 2021-06-08:
I'm able to get the link quality through the XCTU tool, so I would expect it should be possible with the Python.

tatianaleon, 2021-06-08:
Hi @seanhodgson, 

As you pointed out, there was an issue in the code. It is fixed in master (commit b7a9b5eec89ab2d0f312633d71536685257b8e70). amd will be included in the next release.

In the meantime, you can download the source code from the master branch and install the library from there (see instructions at https://github.com/digidotcom/python-xbee/blob/master/README.rst#install-from-source).

Sorry for the inconvenience.

Best regards.

seanhodgson, 2021-06-08:
Understood

seanhodgson, 2021-06-08:
Just out of curiosity, do you have any idea when you might be doing a new release of the xbee-python?

seanhodgson, 2021-06-08:
as an FYI:  pulling from the master branch and installing locally has resolved my issue.  Thank you.