digidotcom/xbee-python #124: When using Xbee S2C Pro Mesh getting AtCommandException

JeffLabonte, 2020-01-31:
Hi there! 

I want to report to you that we are getting a ATCommandException when we open the connection with our XBee Mesh S2C Pro. It has something to do with the SS command that is not implemented in the radio. 

My collegue and I have notice that the SS command is no longer used when it comes to that radio.

Thank you guys!

JeffLabonte, 2020-01-31:
It seems that it was fixed in one of your commit, but you guys haven't pushed the fix to Pypi.

tatianaleon, 2020-02-13:
Hi @JeffLabonte 

As you said, it is already fixed in master (see commit a10f3df) and will be included in the next release.
(issue #103.)

In the meantime, you can download the source code from the master branch and install the library from there (see instructions at https://github.com/digidotcom/python-xbee/blob/master/README.rst#install-from-source).

Sorry for the inconvenience.

Best regards.

closed 2020-02-13 by tatianaleon