digidotcom/xbee-python #158: Use this library with cc6ulsbc pro

pablovi96, 2020-03-24:
Hi,
How do I use this library with a conectcore 6ul sbc pro with yocto proyect? I have already installed the library and i want to communicate with an xbee 3.0 attached to the connectcore but i dont know how.

rubenmoral, 2020-03-24:
Hi @pablovi96,

In order to communicate with the XBee module, you just have to instantiate an `XBeeDevice` object using the `/dev/ttyXBee` port and the baudrate of the module (9600 by default).

```python
from digi.xbee.devices import XBeeDevice

xbee = XBeeDevice("/dev/ttyXBee", 9600)
xbee.open()
...
```

Remember that the XBee must be configured in API mode to work with this library.

You can find more information in the official documentation: https://xbplib.readthedocs.io/en/latest/getting_started_with_xbee_python_library.html.

Regards.

closed 2020-03-24 by rubenmoral

pablovi96, 2020-03-24:
Ok, I understand but I dont have a xbee development board so I cant connect to XCTU and I cant set the API mode (I suppose that the XBee is in transparent mode). I have to buy a development board so I can configure it first, or there is a way to set into API mode through the cc6ul without connecting to XCTU?

rubenmoral, 2020-03-24:
You can configure it directly from the CC6UL without needing to have a dev board. For that, you can use the microcom utility to talk with the module in transparent mode, `microcom -s 9600 /dev/ttyXBee`.

Then, you have to send the following commands to the module (all of them except the first one ending with new line, and waiting for the module to respond with OK between each one):

```
+++       (to enter command mode)
atap1     (to set operating mode to API)
atwr      (to write parameters)
atac      (to apply parameters)
atcn      (to exit command mode)
```

Regards.

pablovi96, 2020-03-27:
Hi again,
I've configured the XBee module into API mode. But when I try to follow the instructions to instantiate an xbeedevice object i have this error:
-sh: from: not found
I'm a begginner so I'm not sure why this error appears. I suppose that if the python library is installed it should work the python interpreter.
What can i do?

rubenmoral, 2020-03-30:
Could you please clarify the issue? How are you running the code? In a script, or in the Python interpreter directly? If in a script, are you invoking Python to run it (`python3 my_script.py`)? If in the interpreter, which sentence causes the error?

pablovi96, 2020-03-30:
I open the python interpreter with: python().
then I do the following:
```
root@ccimx6ulsbc:~# python()
> from digi.xbee.devices import XBeeDevice
root@ccimx6ulsbc:~# python()
> device = XBeeDevice("/dev/ttyXBee", 9600)
-sh: syntax error: unexpected "("
```
So I dont know what the problem is.

rubenmoral, 2020-03-30:
Well, that's not the way of running a Python application. You can either create a .py script with all the code and then run it by invoking it (`python3 my_script.py`), or, if you want to quickly test the library like in your case, open the Python interpreter and run there the code:

```
root@ccimx6ulsbc:~# python3
Python 3.6.10 (default, Jan 10 2020, 13:28:46)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from digi.xbee.devices import XBeeDevice
>>> device = XBeeDevice("/dev/ttyXBee", 9600)
>>> device.open()
>>> ...
```

Hope this helps.

pablovi96, 2020-03-30:
ok.
Thank you very much!