digidotcom/xbee-python #164: How does the xbee.send_data() work for HA messages?

pablovi96, 2020-04-02:
Hi I want to send a message with an xbee3 zigbee with this python library to a Home automation device.
The thing is that I dont know what the message should be, because i dont know how the function works. I have to send all the API frame from length to payload bytes as in XCTU (with theframe generator) or I can put only the bytes corresponding to the cluster ID and payload?

Thank you in advance

rubenmoral, 2020-04-02:
Hi @pablovi96,

You have to use the appropriate methods to send explicit data, see this chapter: https://xbplib.readthedocs.io/en/latest/user_doc/communicating_with_xbee_devices.html#send-and-receive-explicit-data.

Hope this helps.
Regards.

closed 2020-04-02 by rubenmoral

pablovi96, 2020-04-02:
Ok, this seems to work but when I send the same order as in the example the following error occurs:
```
Traceback (most recent call last):
  File "etc/init.d/udp.py", line 45, in <module>
    xbee.send_expl_data(remote,1,1,clid,prfid, message)
  File "/usr/lib/python3.5/site-packages/digi/xbee/devices.py", line 3059, in send_expl_data
    profile_id, transmit_options)
  File "/usr/lib/python3.5/site-packages/digi/xbee/devices.py", line 1028, in dec_function
    return func(self, *args, **kwargs)
  File "/usr/lib/python3.5/site-packages/digi/xbee/devices.py", line 1038, in dec_function
    response = func(*args, **kwargs)
  File "/usr/lib/python3.5/site-packages/digi/xbee/devices.py", line 2133, in _send_expl_data
    False, transmit_options))
  File "/usr/lib/python3.5/site-packages/digi/xbee/devices.py", line 2556, in __build_expldata_packet
    cluster_id, profile_id, 0, transmit_options, data)
  File "/usr/lib/python3.5/site-packages/digi/xbee/packets/common.py", line 2091, in __init__
    if dest_endpoint < 0 or dest_endpoint > 255:
TypeError: unorderable types: bytes() < int()
```
I thougth that I could change the inputs from bytes to int but didnt work. 
I have put it like this also:
```
se=b'\x01' #source endpoint
de=b'\x01' #destination endpoint
clid=b'\x00\x06 ' #cluster id
prfid=b'\x01\x04' #profile id
```

any idea of how should I put the function inputs?

rubenmoral, 2020-04-02:
They must be integers:

```
xbee.send_expl_data(remote, 0x01, 0x01, 0x0006, 0x0104, message)
```

Please take a look at this example: https://github.com/digidotcom/xbee-python/blob/master/examples/communication/explicit/SendExplicitDataSample/SendExplicitDataSample.py.

pablovi96, 2020-04-02:
Ok the problem was that I was trying this:
`xbee.send_expl_data(remote, 0x01, 0x01, 0x0006, 0x0104, message)`
and as in the example i need to put in this order:
`xbee.send_expl_data(remote, message, 0x01, 0x01, 0x0006, 0x0104)`

Thank you very much!