digidotcom/xbee-python #179: Create an example code snippet to show how to send an API frame to a remote XBee.

securitybryan, 2020-06-05:
All the examples send a trivial "Hello World!" string to a remote XBee.

```
DATA_TO_SEND = "Hello XBee!"
device.send_data(remote_device, DATA_TO_SEND)
```
It would be great to show something a bit more practical such as how to send an API frame.  I have slogged through the docs looking for an example, and, judging by all the posts on StackOverflow this is a very common question.

tatianaleon, 2020-06-08:
Hi @securitybryan,

When you communicate with an XBee (your local device or any remote), the library is already sending API frames. API frames are used for any operation with your XBee devices: local and remote configuration, data transmission/reception to/from a remote, network discovery, getting IO samples, etc.

In the example above it is sending a "0x10 Transmit Transmit request" to `remote_device`. If `remote_device` is the coordinator, it could be something similar to:

7E 00 19 10 0C 00 00 00 00 00 00 00 00 00 00 00 00 48 65 6C 6C 6F 20 58 42 65 65 21 4A

The goal of the API is to allow configuration, communication, and work with XBee devices without the necessity of understanding the API frames low level. This is why, all the examples and the documentation explain operations and not raw API frames.

If you need to see sent and received frames when an operation is performed, you can enable debugging using the following in your code:
```
utils.enable_logger("digi.xbee.devices", logging.DEBUG)
utils.enable_logger("digi.xbee.reader", logging.DEBUG)
``` 

Best Regards.

securitybryan, 2020-06-08:
@tatianaleon Thank you for clarifying this point.  I must have missed it in my haste.  Do the remote devices then need to be in API mode to receive these commands?  E.g.

`device.send_data(remote_device, DATA_TO_SEND)`

securitybryan, 2020-06-08:
As an aside, I'd love to see someone on this project monitor StackOverflow for questions regarding this library.  I know personally I would have adopted it much soon (and with much more success) had someone on SO been able to answer fairly fundamental questions.

tatianaleon, 2020-06-09:
Hi @securitybryan,

First, we are doing our best trying to solve questions :-)

Regarding the operating mode of a remote node, XBee devices will always receive wireless data addressed to it, independently of its operating mode (of course, they must be in the same network). This configuration "only" establishes the way a host communicates with an XBee module through the serial interface, and not how XBee nodes communicate over the air.

But, if you want **to use this library on the remote node side, it must be in API mode** (the library only supports API modes).

I don't know which XBee module are you using, but my recommendation is to read its user guide. You can find the different XBee radios at https://www.digi.com/products/embedded-systems/digi-xbee/rf-modules. Once you select your hardware, under the "Product Support" tab you will find a link to the user guide.

A more detailed explanation of operating modes, including their differences and limitations, can be found at [Operating modes](https://www.digi.com/resources/documentation/Digidocs/90001942-13/Default.htm#concepts/c_serial_communication.htm).

As a brief summary:

Transparent and API modes represents the way you "talk" with the XBee via the serial interface. I mean, the "language" used to command the XBee to send data to a remote or to notify about received data from any remote. We can say there are 2 communication processes when an XBee sends data to another XBee:
* **Serial communication**, between the XBee and the host connected to the serial interface. This communication can be set to use API or Transparent mode, and depending on this configuration, commands and data must follow one format or another.
* **Wireless communication**, between XBee modules. This is always the same process regardless of the configured operating mode (API or Transparent).

[image]

The difference between API and Transparent operating modes is how the information is presented through the serial port:
* Transparent mode: The radio passes information along exactly as it receives it.
[image]

* API mode: Data is communicated in packets (commonly called API frames)
[image]

Best Regards.

securitybryan, 2020-06-09:
@tatianaleon   Please know that wasn't meant as a slam  :-)    I would love to be able to hang out where other XBee developers are and that doesn't seem to be SO.  But, as a developer, I don't want to open issues for the sake of enablement either.

I thought I was pretty knowledgeable about XBees after many years of C programming for them.  But, your post was very helpful.   I fully understand how API mode works which is why asked about it being required.  I must have come across a little too ignorant...  Thank for all your work on this!  I'll go ahead and close out this issue.

closed 2020-06-09 by securitybryan

tatianaleon, 2020-06-09:
@securitybryan, there are quite a lot of people using XBee modules and confused about how operating modes work... I thought it was your case... I see probably you know much more than me about that, I did not know you were an old colleague :-)
Glad to be helpful. Thanks @securitybryan.

securitybryan, 2020-06-09:
@tatianaleon  I seriously doubt that I know more.  I just muddle along as best I can ;-)