digidotcom/xbee-python #183: set_node_id() is updating multiple remote nodes

aarontusko, 2020-07-01:
I have a network with 3x XBee3 modules running firmware 0x200A.
I am using the XBee Python library version 1.3.0.
 
When I try to update a remote modules node id, it updates both remote modules.
Here is a sample python interactive session demonstrating the bug.

pi@pitouch:~ $ python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
>>> 
>>> 
>>> # Import 
>>> from digi.xbee.devices import XBeeDevice
>>> 
>>>
>>>
>>> # Setup XBee Device
>>> xbee = XBeeDevice( '/dev/xbee3_api', 115200 )
>>> xbee.open()
>>>
>>> 
>>>
>>> # Setup / Discover Network
>>> xnet = xbee.get_network()
>>> xnet.start_discovery_process()
>>> 
>>>
>>>
>>> # Get Remote Modules
>>> a = xnet.get_devices()[0]
>>> b = xnet.get_devices()[1]
>>> 
>>>
>>>
>>> # Show Module 'a' information
>>> a.get_64bit_addr().address.hex()
'0013a20041b680a6'
>>> a.get_node_id()
'Bob'
>>> 
>>>
>>>
>>> # Show Module 'b' information
>>> b.get_64bit_addr().address.hex()
'0013a2004182c484'
>>> b.get_node_id()
'Jay'
>>> 
>>>
>>>
>>> # Change node id on module a
>>> a.set_node_id('Susanne')
>>> 
>>>
>>>
>>> # Show that module 'a' has updated node id
>>> a.get_node_id()
'Susanne'
>>>
>>> 
>>>
>>> # Module 'b' looks the same until...
>>> b.get_node_id()
'Jay'
>>> 
>>>
>>>
>>> # Now module 'b' shows the incorrect node id.
>>> b.read_device_info()
>>> b.get_node_id()
'Susanne'
>>> 
>>> 
>>>

tatianaleon, 2020-07-03:
Hi @aarontusko,

Did you configure different 16-bit addresses for your modules (`MY` parameter)? Or all of them to `0xFFFE`?
In 802.15.4, to use long 64-bit address, `MY` must be `0xFFFE` or `0xFFFF`, otherwise modules use short source addressing (`MY` address).
By default, `MY` is `0x0000` in 802.15.4 firmware, so, by default, modules use short 16-bit addressing and `0x0000` as address. This means, when sending a Remote AT Command to `0x0000`, all the remote nodes in range will receive it and then change their `NI`.

See https://www.digi.com/resources/documentation/digidocs/90002273/Default.htm#reference/r_addressing_802.htm

Best Regards,
Tatiana

aarontusko, 2020-07-06:
Hello Tatiana,

I just ran the test again setting MY to 0xFFFE on all three modules and everything works as expected.
In my previous test I had them all set to 0xFFFF which I thought would be the same as setting them to 0xFFFE. 
I will use 0xFFFE from now on. 

Thank You, 
-Aaron

closed 2020-07-06 by aarontusko