digidotcom/xbee-python #184: Dicovery not working

sami-baroudi, 2020-07-02:
Hello,
I'm facing a problem with discovery.
I'm using a S2C in Coordinator mode + API2.
When launching the discovery, it returns an empty list. But XCTU detects the other device.
I tried factory resetting the device, but still it returns an empty list.

Code snippet :
[image]

tatianaleon, 2020-07-03:
Hi @sami-baroudi,

I see in your code that you are getting the list of devices with the instruction:

```
device_list = self.get_devices()
```

Shouldn't it be

```
device_list = xbee_network.get_devices()
```

Best Regards,
Tatiana

sami-baroudi, 2020-07-06:
Hi @tatianaleon,

Thank you for your reply.

In my code I implemented a get_devices method, that does the same thing as the xbee_network.get_devices() method.

here is my method : 
[image]

Best regards, 
Sami

securitybryan, 2020-07-06:
Have you tried using a longer timeout?  I am not able to get a list of devices in 5 seconds on my network.

sami-baroudi, 2020-07-06:
hello @securitybryan,

5 seconds is only a default value, but I usually use a timeout above 30 seconds.
Still, no device discovered.

If the device was discovered by xctu, and the pairing was done by xctu itself will it stop the discovery process in python from detecting the device?

tatianaleon, 2020-07-06:
Hi @sami-baroudi,

It is difficult to know what is going on when the code running is unknown.
Does the _[DiscoverDevicesSample.py](https://github.com/digidotcom/xbee-python/blob/master/examples/network/DiscoverDevicesSample/DiscoverDevicesSample.py)_ example in the library work for you?

Best Regards,
Tatiana

securitybryan, 2020-07-06:
Hi @sami-baroudi ,

No, you should still be able to discover the devices from your Python code.

sami-baroudi, 2020-07-06:
Hi @tatianaleon,

I already used [DiscoverDevicesSample.py](https://github.com/digidotcom/xbee-python/blob/master/examples/network/DiscoverDevicesSample/DiscoverDevicesSample.py) example before implementing any of my methods.

The result is : 

> Discovering remote XBee devices...
>Device discovered: 0013A20041AE4A0E -  
>Discovery process finished successfully.

the device discovered is the module I use (XBEE s2c) to pair with other devices, but besides the module itself the discovery doesn't display that it found my device, and the xbee_network.get_devices() returns an empty list.

tatianaleon, 2020-07-06:
Hi @sami-baroudi,

Is the other device not found an end device? I mean, is it sleeping? If so, how much time is awaken and sleeping?
How do you discover the remote node in XCTU, using the "Discover radio nodes in the same network" button of the local XBee or with the Network working mode?

Best Regards,
Tatiana

sami-baroudi, 2020-07-07:
hi @tatianaleon,

to look for my device, I follow this steps:
1 - I switch to network working mode
2 - I scan the radio module network
3 - It automatically shows me the devices that are in my network

I added a snippet for more info.
[image]

Thank you for your response.

tatianaleon, 2020-07-07:
Hi @sami-baroudi,

So, your remote device does not sleep since it is a router and you are using the Network working mode.
The discovery mechanism used in this mode and the one you are using in the with the Python library are different.

To invoke the same mechanism in the XCTU you must use the **Discover radio nodes in the same network** button (see [Discover remote radio modules](https://www.digi.com/resources/documentation/digidocs/90001458-13/default.htm#task/t_discover_remote_radio_modules.htm)). This is invoking the `ND` command to discover nodes in the network. **Use this button to compare the results with your current code results**.

Another option is to use the XCTU Network working mode mechanism in the library. In this case, you must use `ZigbeeDevice` and `RemoteZigbeeDevice`objects (instead of `XBeeDevice` and `RemoteXBeeDevice`) and invoke a "deep" discovery. This is asking for the routing and neighbor tables of each Zigbee node. Here you have a code snippet:
```python
xbee = ZigBeeDevice(PORT, BAUD_RATE)
xbee.open()
xnet = device.get_network()

xnet.start_discovery_process(deep=True)
while xnet.is_discovery_running():
    time.sleep(1)

print("Current network:\n%s" % '\n'.join(map(str, xnet.get_devices())))
```

Best Regards,
Tatiana

sami-baroudi, 2020-07-07:
Hi @tatianaleon,

Indeed, the **Discover radio nodes in the same network** button doesn't return devices in XCTU.

But luckily, the deep discovery returns the node I'm looking for, so thanks a lot...big relief.

**One issue that you need to check:** When installing the digi xbee package with pip, the start discovery process does not support the deep parameter even though the version installed is 1.3.0. Because the method itself does not support any other parameter.
I had to clone the repository and install it, to get the modified version of **start_discovery_process()**.
Maybe other methods are not updated when installing with pip install. Need to check that.

Anyway, thanks a lot @tatianaleon.

Best regards,
Sami

securitybryan, 2020-07-08:
This is possibly the most useful issue thread I have seen in a long time.    The documentation is very comprehensive but it is hard (for me at least) to find this kind of information. 

Thank you!

tatianaleon, 2020-07-09:
Sorry @securitybryan, you are right, I forgot to tell you that the "deep" discovery functionality is only available in master and not released yet.

securitybryan, 2020-07-09:
No worries @tatianaleon !  I am hoping to be able to contribute to the docs and examples at some point.  I need a few more miles behind me working with this library first...

closed 2020-07-21 by sami-baroudi