digidotcom/xbee-python #204: Python 3.9 SyntaxError

souellet-ms, 2020-10-13:
Using digi-xbee with Python 3.9 (release October 5th, 2020) gives the following error. digi-xbee works fine with Python 3.8

Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "C:\Users\Simon\Documents\motsai\P175\test\xbee_bandwidth\venv\lib\site-packages\digi\xbee\devices.py", line 22, in <module>
    from digi.xbee import serial
  File "C:\Users\Simon\Documents\motsai\P175\test\xbee_bandwidth\venv\lib\site-packages\digi\xbee\serial.py", line 21, in <module>
    from digi.xbee.models.mode import OperatingMode
  File "C:\Users\Simon\Documents\motsai\P175\test\xbee_bandwidth\venv\lib\site-packages\digi\xbee\models\mode.py", line 230
    return sum(op.code for op in options if lambda option: (option != cls.EXPLICIT))
                                                                   ^
SyntaxError: invalid syntax
python-BaseException

tirkarthi, 2020-10-14:
This is due to new parser which was closed as not fixed in https://bugs.python.org/issue41848 . lambda is always evaluated as true so this is a bug that needs to be fixed in the code.

tatianaleon, 2020-10-15:
Hi @souellet-ms,

Thanks for reporting this issue. We have just fixed it in `master` (see commit https://github.com/digidotcom/xbee-python/commit/ddbff360c0553b30fac1e7d97c1dfbd1b99db09b) and will be included in the next release.

In the meantime, you can download the source code from the master branch and install the library from there (see instructions at https://github.com/digidotcom/python-xbee/blob/master/README.rst#install-from-source).

Sorry for the inconvenience.

Best regards.

closed 2020-10-15 by tatianaleon

traverseda, 2021-03-08:
When is the next release? Waiting on this fix.