digidotcom/xbee-python #209: New release and changelog

FRVisser, 2020-11-05:
Hello Digi, 

The library has seen a lot of commits. It seems good to release a new (sub)version and update the changelog. As the 'summer update' as mentioned in our calls is not seen back in the releases.

Thanks,
Fokko

duysqubix, 2021-01-08:
bumping this - I haven't noticed the massive amounts of commits since 1.3.0. Was tracking down something only to discover src from 1.3.0 is vastly different

duysqubix, 2021-02-15:
@brandonmoser who can we tag to get this issue some attention? I am about at the point to fork this project and create my own release to upload to pip so I can use in projects instead of pulling from git directly.

FRVisser, 2021-02-15:
I'll also send I mail to my contact at Digi

brandonmoser, 2021-02-15:
> I'll also send an email to my contact at Digi

Any support request should go to Tech.Support@digi.com <mailto:Tech.Support@digi.com>.

duysqubix, 2021-02-15:
@brandonmoser who is the owner of this repo?

brandonmoser, 2021-02-15:
Digi International (https://www.digi.com/ <https://www.digi.com/>) is the owner. One of the engineering teams supports the project.

duysqubix, 2021-02-26:
@brandonmoser is there a direct contact for one of the engineering team members? Whether via git or email?

tatianaleon, 2021-03-19:
Hi @FRVisser,

Version 1.4.0 is released and available in PyPI: https://pypi.org/project/digi-xbee/1.4.0/

Best Regards

closed 2021-03-19 by tatianaleon