digidotcom/xbee-python #210: Error processing packet -Wrong Checksum

aabs7, 2020-11-22:
I am trying to send API frame with payload through Xbee pro S3B. It works fine, but it takes a lot of time (4s - 8s) to send the payload when checksum error occurs. The payload that I'm sending is way below the limit of xbee that I'm using i.e 256 bytes.
In the log, the error looks as follows-

```
2020-11-22 06:11:47,401 : ERROR :: Error processing packet '7E 00 07 8B 4E FF 7E 00 00 00 29': Wrong checksum
2020-11-22 06:11:56,065 : ERROR :: Error processing packet '7E 00 07 8B 57 BF FE 00 00 00 20': Wrong checksum
2020-11-22 06:12:01,814 : ERROR :: Error processing packet '7E 00 07 8B 5B BF FE 00 00 00 1C': Wrong checksum
```
I need to send data very fast, but the checksum error is causing bit of a issue.

aabs7, 2020-11-22:
The issue is when I run the code in raspberry pi. The code works perfectly fine in laptop.

tatianaleon, 2020-11-23:
Hi @aabs7,

I guess you are using the `send_data` method of the API, if so and if the received transmit status is corrupted (checksum is invalid) the library waits the configured timeout for a valid one. In fact, the problem is why the library is receiving corrupted frames.

I'm not sure what you mean with "I need to send data very fast", I will assume you are using a high baudrate. When working at high baudrates ensure hardware flow control is enabled on both sides of the serial connection: the XBee and the Raspberry in this case.

Best Regards,
Tatiana

aabs7, 2020-11-29:
Hello @tatianaleon ,
I am using ```send_data``` method to send api frame with drone status as payload. When I run the code in my laptop then everything works fine, checksum error doesn't occur. However, When I run the code on raspberry pi, checksum error occurs frequently. 

I have decreased the timeout to 1 second in the API which made the data be sent every one second but I still don't know why the checksum error occurs (**in raspberry pi**).  Do you know why XBee is receiving corrupted frames in raspberry pi?

Thank you.

tatianaleon, 2020-11-30:
Hi @aabs7,

I will assume you are using a high baudrate in the serial connection. When working at high baudrates **ensure hardware flow control is enabled on both sides of the serial connection**: the XBee and the Raspberry in this case.

Best Regards,
Tatiana

aabs7, 2020-12-01:
Hello @tatianaleon ,
After enabling hardware flow control, Do I have to connect XBee in the GPIO pins of raspberry pi? or connection through USB works too?

aabs7, 2020-12-15:
The error is resolved when interfacing XBee using GPIO (rx,tx) of raspberry pi and enabling hardware flow control in both XBee and Raspberry Pi. Thank you @tatianaleon

closed 2020-12-15 by aabs7