digidotcom/xbee-python #214: Firmware update issue with firmware names with hexadecimal naming

JoonaMorsky, 2020-12-15:
in file digi/xbee/profile.py in update_firmware function line 802 self._firmware_version = int(firmware_element.get(_XML_FIRMWARE_FIRMWARE_VERSION)) tries to parse firmware names as base 10 when newer versions of firmware uses base 16 naming scheme. (For example 100A, 100B ...)

Traceback :
    xbee_profile = XBeeProfile(PROFILE_FILE)
   File "/usr/local/lib/python3.7/dist-packages/digi/xbee/profile.py", line 653, in __init__
     self._parse_xml_firmware_file()
   File "/usr/local/lib/python3.7/dist-packages/digi/xbee/profile.py", line 802, in _parse_xml_firmware_file
     self._firmware_version = int(firmware_element.get(_XML_FIRMWARE_FIRMWARE_VERSION))
ValueError: invalid literal for int() with base 10: '100A'

Device: Digi XBEE Zigbee 3.0 XB3-23 series

tatianaleon, 2020-12-15:
Duplicates https://github.com/digidotcom/xbee-python/issues/177

tatianaleon, 2020-12-15:
Hi @JoonaMorsky,

Thanks for reporting this issue. There is already a fix in master (see commit [814c5e9](https://github.com/digidotcom/xbee-python/commit/814c5e999a4752a121d0a6b445c5d1eead3a862e)) and will be included in the next release.

If you have problems working with the source code in master, please let us know.

Sorry for the inconvenience.

Best regards.

closed 2020-12-15 by tatianaleon