digidotcom/xbee-python #215: Recovery after manual reset (pin RESET to low)

ibantxo, 2020-12-15:
Hi,

I have launch my "sender" program. That is a loop that sends 1 message, waits 10 seconds, and sends another message....
Then, when I put RESET pin to 0, the chip then resets correctly.

The "modem_status_received_callback" function detects --> [0] "Device was reset"
How could I recover the communication via serial with the chip?
I have tried: device.reset(), device.close(), device.open(), del device... but no working...
The program logs some of the next messages, depending on code changes:
- [Transmit Exception] Attempting to use a por that is not open
- [Transmit Exception]There was a problem with the transmitted packet response (status not ok)
- [XbeeException] XBee device's serial port closed.

Only exit the python program and launch again solves the problem.

How could I recover the communication without exit the program?

Many Thanks in advance

Iban

tatianaleon, 2020-12-16:
Hi @ibantxo,

What is managing the RESET pin? the same application?
Could you send a code example where this issue is reproducible?

Best Regards

ibantxo, 2020-12-16:
Hi @tatianaleon !!

Thank you for the response.
Yes, the RESET (hardware) is an input pin (for example, the 6th in the Xbee SX 868). "Drive low to reset device". (min 100ns)

I am writting a script to work 24x7. With a frequency of 1/10s the 'TX node' sends a message to 'RX node'. Everything was OK, but sometimes the system stop working: the script goes on with its infinite loop, but the device.reset() does not working. So... I decided to try a "hard RESET" (using the pin). A 0V in tha pin resets the Xbee with no problem, but my python program cannot connect again. The program detects the reset via  "modem_status_received_callback" function detects --> [0] "Device was reset". I have tried to reconnect but I have not got it.

When this happens, if I stop the script and launch it again, everything works great again. So the target is to ensure I can reset the hardware and the communication to continue sending messages.

This is the TX script:
`# Copyright 2017, Digi International Inc.
#
# Permission to use, copy, modify, and/or distribute this software for any
# purpose with or without fee is hereby granted, provided that the above
# copyright notice and this permission notice appear in all copies.
#
# THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES
# WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF
# MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR
# ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES
# WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN
# ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF
# OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.
from digi.xbee.devices import XBeeDevice
#from digi.xbee.packets.base import DictKeys
from digi.xbee.exception import XBeeException, TimeoutException, InvalidOperatingModeException, ATCommandException, OperationNotSupp>
from serial import Serial, SerialException
from subprocess import call
from digi.xbee.devices import XBeeDevice
from digi.xbee.util import utils
from digi.xbee.models.address import XBee64BitAddress
from digi.xbee.models.status import PowerLevel,ModemStatus

import datetime
import json
import random
import os
import time

# TODO: Replace with the serial port where your local module is connected to.
PORT = "/dev/ttyS0"
# TODO: Replace with the baud rate of your local module.
BAUD_RATE = 9600

PARAM_VALUE_PAN_ID = bytearray(b'\x01\x23')
PARAM_DESTINATION_ADDR = XBee64BitAddress.BROADCAST_ADDRESS
PARAM_POWER_LEVEL = PowerLevel.LEVEL_MEDIUM

PARAM_NODE_ID = "NI"
PARAM_PAN_ID = "ID"
PARAM_DEST_ADDRESS_H = "DH"
PARAM_DEST_ADDRESS_L = "DL"

DATA_TO_SEND = "Hello XBee!"
REMOTE_NODE_ID = "LOCAL_DEVICE"

def main():
    print(" +-----------------------------------------------------+")
    print(" | XBee Python Library Send Data Asynchronously Sample |")
    print(" +-----------------------------------------------------+\n")

    device = XBeeDevice(PORT, BAUD_RATE)

    # Define the callback.
    def my_modem_status_callback(status):
        print("Modem status: [%s] %s" % (status.code,status.description))
        if(status.code==0):
           if device is not None and device.is_open():
              print ("Detected as open")
              device.close()
[... 72 lines cut ...]

#    except XBeeException as inst:
#        print ("#########################XBeeException################################")
#        print (type(inst))
#        #print (inst.args)
#        #print (inst)
#        print("ERROR: %s" % str(inst))
#        #return

    except TransmitException as e:
        print ("#########################XBee TransmitException################################")
        print("ERROR TIMEOUT: %s" % str(e))
        return

    finally:
        if device is not None and device.is_open():
            device.close()

if __name__ == '__main__':
    main()
`

Many thanks in advance

Iban

tatianaleon, 2021-03-19:
Hi @ibantxo:

In your example, I cannot see anything resetting the XBee (nor software nor hardware reset) I only see the function `my_modem_status_callback` that is not registered.

```python
    # Define the callback.
    def my_modem_status_callback(status):
        print("Modem status: [%s] %s" % (status.code, status.description))
        if status.code == 0:
            if device is not None and device.is_open():
                print("Detected as open")
                device.close()
```

This function is closing the connection with the XBee and it is never opened again, so, in case it is registered, things are not going to work: the next time the application tries to send anything, it will throw an exception because the connection is closed.

I do not think that a software reset (`device.reset()`) or a hardware reset (Pin 6 RESET, https://www.digi.com/resources/documentation/Digidocs/90001538/#reference/r_pinouts.htm) have a high impact in your application (the boot time of the XBee firmware application is really short).

Best Regards

closed 2022-02-16 by tatianaleon