digidotcom/xbee-python #221: BUG python 3.8.7

FRVisser, 2021-01-21:
Using the latest from the master branch I hit an import error on python  3.8.7. Swiching back to python 3.7 resolved my issue. 

```
digi/xbee/packets/factory.py", line 108, in <module>
cannot import name 'FSCmd' from partially initialized module 'digi.xbee.models.filesystem' (most likely due to a circular import)
```

tatianaleon, 2021-01-22:
Hi @FRVisser,

Can you let us know what application are you trying to run?

Best Regards

FRVisser, 2021-01-22:
I have a unit test system which uses the `digidotcom/xbee-python` `factory` function to interpreted packages withoud the library. I can reproduce the issue as follows:

```python
Python 3.8.7 (v3.8.7:6503f05dd5, Dec 21 2020, 12:45:15) 
[Clang 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from digi.xbee.packets import factory
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/***MyProjectFolder**/digi/xbee/packets/factory.py", line 108, in <module>
    from digi.xbee.packets.filesystem import FSRequestPacket, FSResponsePacket, RemoteFSRequestPacket, RemoteFSResponsePacket
  File "/***MyProjectFolder**/digi/xbee/packets/filesystem.py", line 15, in <module>
    from digi.xbee.models.filesystem import FSCmd, FSCmdType, OpenFileCmdRequest, \
  File "/***MyProjectFolder**/digi/xbee/models/filesystem.py", line 15, in <module>
    import digi.xbee.filesystem
  File "/***MyProjectFolder**/digi/xbee/filesystem.py", line 31, in <module>
    from digi.xbee.models.filesystem import FSCmd, GetPathIdCmdRequest, \
ImportError: cannot import name 'FSCmd' from partially initialized module 'digi.xbee.models.filesystem' (most likely due to a circular import) (/***MyProjectFolder**/digi/xbee/models/filesystem.py)
>>>

```

tatianaleon, 2021-01-28:
Fixed in master (commit https://github.com/digidotcom/xbee-python/commit/85deb30c08c2200d97cb67062f67bcafe86b864a)

closed 2021-01-28 by tatianaleon