digidotcom/xbee-python #222: read failed after minutes of running

jrleeman, 2021-01-22:
On Ubuntu 20.04 with Python 3.8 after continuously running and receiving data for a few minutes we get the message

`read failed: device reports readiness to read but returned no data (device disconnected or  multiple access on port?)`

with a traceback starting at line 444 in `xbee/reader.py` and going all the way down to `serial/serialposix.py`. The only solution appears to be to kill the process and start it again, but that's obviously non-ideal. It's not a hardware issue as XCTU reads continuously for hours with no issue. 

Nothing fancy happening in my code:

```python
def process_xbee_packet(xbee_message):
    """
    Processes the xbee packet and writes it to file.
    """
    well_data = {}
    raw_packet = xbee_message.data
    well_data['time'] = datetime.utcnow()
    well_data['start_byte'] =raw_packet[0]
    well_data['transmitter_id'] = raw_packet[1]
    well_data['depth'] = struct.unpack('f', raw_packet[2:6])[0]
    well_data['water_temperature'] = struct.unpack('f', raw_packet[6:10])[0]
    well_data['air_pressure'] = struct.unpack('f', raw_packet[10:14])[0]
    well_data['air_temperature'] = struct.unpack('f', raw_packet[14:18])[0]
    well_data['battery'] = struct.unpack('H', raw_packet[18:20])[0]
    well_data['end_byte'] = raw_packet[20]
    print(well_data['transmitter_id'])

    # Write it to file if the packet is complete (start and end bytes are correct)
    if (well_data['start_byte'] == 190) and (well_data['end_byte'] == 239):
        write_well_data(config, well_data)

if __name__ == '__main__':
    try:
        device = XBeeDevice(config['XBEE']['ComPort'], config['XBEE']['BaudRate'])
        device.open()
        device.add_data_received_callback(process_xbee_packet)
        input()
    except (KeyboardInterrupt, SystemExit):
        device.close()
```

Artyrm, 2021-08-20:
Someone is accessing your serial port. Lib just bails out in such cases, but you can try to handle it my way: https://github.com/digidotcom/xbee-python/issues/252#issuecomment-901416300

tatianaleon, 2022-02-16:
Solved in [1.4.1](https://github.com/digidotcom/xbee-python/releases/tag/1.4.1).

closed 2022-02-16 by tatianaleon