digidotcom/xbee-python #237: Hi @arunnediyasala,

mohamadnoor991, 2021-04-19:
Hi @arunnediyasala,

This error means that Python cannot find the pyserial library, which is needed for the serial communication. In theory it should have been installed automatically when installing the XBee Python library, that's weird...

Anyway, you can install it manually:

`pip install pyserial`

Regards.

_Originally posted by @rubenmoral in https://github.com/digidotcom/xbee-python/issues/1#issuecomment-341343521_

tatianaleon, 2021-04-19:
Hi @mohamadnoor991,

This is a Python library that cannot run inside and XBee 3.
XBee 3 runs MicroPython and pip is not working there. Please refer to the [Digi MicroPython Guide](https://www.digi.com/resources/documentation/digidocs/90002219/default.htm) to know what is available. 

Best Regards

mohamadnoor991, 2021-04-19:
ok that is clear but how i can install the missing module in xbee3 like "UART and digi" ?
i need these modules to start programming and execute my ideas

tatianaleon, 2021-04-19:
Hi @mohamadnoor991,

As far as I know, you cannot install new modules on a XBee 3. Inside the [Digi MicroPython Guide](https://www.digi.com/resources/documentation/digidocs/90002219/default.htm) you can check what is available.

Anyway, the right place to ask about MicroPython on a XBee 3 is https://github.com/digidotcom/xbee-micropython.

Best Regards.

mohamadnoor991, 2021-04-19:
Dear @tatianaleon  thanks for your answer

closed 2021-04-28 by tatianaleon