digidotcom/xbee-python #240: Frame ID value with send_data_async

goowza, 2021-05-20:
Hello,

I have a question regarding the `send_data_async` method. This method is supposed to be used to send some data without waiting for a response packet from the receiver. 

However the value of the frame ID in the packet built using the `send_data_async` method is computed like this in *devices.py* :

```python
if self.get_protocol() == XBeeProtocol.RAW_802_15_4:
    packet = TX64Packet(self.get_next_frame_id(), x64addr,
                        transmit_options, rf_data=data)
else:
    packet = TransmitPacket(self.get_next_frame_id(), x64addr,
                            XBee16BitAddress.UNKNOWN_ADDRESS, 0,
                            transmit_options, rf_data=data)
```
 Maybe I got it wrong, but it doesn't seem like `self.get_next_frame_id()` is ever equal to 0. That means the receiver will still send a response packet, but the emitter will ignore it ? Why ?

Hope someone can enlighten me on this,
Best regards

tatianaleon, 2021-05-20:
Hi @goowza,

As you said the method `send_data_async` sends data and does not wait for a response packet. But this does not mean the transmit status will not be received. As it is documented, the method just does not wait for the transmit status.
If you need to wait for the status, you can use `send_data` method.

Best Regards

goowza, 2021-05-21:
Hi @tatianaleon thanks for your answer.

I understand, but why not disabling the response packet completely by forcing the value 0 in the frame ID byte of the frame ? (Like it is suggested in the [frame examples](https://www.digi.com/resources/documentation/Digidocs/90001942-13/reference/r_zigbee_frame_examples.htm?TocPath=XBee%20API%20mode%7C_____4)).

If the method does not wait for the response packet anyway, why send one ?

Best Regards

tatianaleon, 2021-05-24:
Hi @goowza,

You can register a callback to receive the transmit status packet and do not wait for it when sending data:

```python
xbee = XBeeDevice("/dev/ttyUSB0", 9600)
xbee.open()
xbee.add_packet_received_callback(my_packet_received_callback)

[...]

remote = ...

[...]

xbee.send_data_async(remote, "Hi there!")

[...]

def my_packet_received_callback(packet):
    print("Received packet: %s" % utils.hex_to_string(packet.output()))

[...]
```

Best Regards

closed 2021-05-25 by goowza