digidotcom/xbee-python #241: [FEATURE REQUEST] Update module to API1 or API2 through library

brandonmoser, 2021-05-21:
The library should have a function to set the API mode and not require XCTU (or other serial connection). If, during `device.open()`, it is found to be in `AP0` (transparent) mode, automatically switch the mode so that all other commands work, as expected.

tatianaleon, 2021-05-24:
Hi @brandonmoser, 

The library already includes a `force_settings` option in the `open()` method to establish the serial connection with the specified serial settings, configuring the XBee with the selected serial values, and setting the operating mode to API without escapes (AP=1). For example, to connect with the XBee attached to the port `/dev/ttyUSB0` using a baud rate of 115200, use:
```python
xbee = XBeeDevice(port="/dev/ttyUSB0", baud_rate=115200)
xbee.open(force_settings=True)
```
This configures the XBee to work in API without escapes (AP=1) and a baud rare of 115200 (BD=7) if it is not already, then opens the serial connection.

Best Regards.

brandonmoser, 2021-05-24:
thanks @tatianaleon. I didn't see this in the Getting Started doc, only that I would need to use XCTU to set the AP mode. Can we note that the Getting Started guide should note this functionality?

tatianaleon, 2021-06-04:
Hi @brandonmoser,

For the get started process there are more things to configure apart from the mode of an XBee, for example, to create a network or connect to an access point. So, for the moment, we are going to leave it as it is. But, we can add some information for the connect method at:

https://xbplib.readthedocs.io/en/latest/user_doc/working_with_xbee_classes.html#open-the-xbee-device-connection

brandonmoser, 2021-06-04:
Thanks Tatiana. I think anything we can do to make the Getting Started process simpler for developers is great.

tatianaleon, 2021-06-08:
Hi @brandonmoser, 

We added some information about the `open()` method in the documentation. For the moment, closing this issue.

Best Regards.

closed 2021-06-08 by tatianaleon