digidotcom/xbee-python #245: Forgotten 802.15.4 bug in node discover options?

Jawkawmaw, 2021-06-12:
Hello

During a small semester project involving XBee devices (XBee S2C), I ran into problems getting the nodes to discover themselves when using 802.15.4. When using the DiscoveryOptions.DISCOVER_MYSELF setting, there are no changes in what is discovered; the device will only discover nodes around it. Conversely, when using the add_device_discovered_callback, the option seems to be automatically enabled and work just fine, yet nothing in the documentation seems to mention this. 

I tried to figure out what the problem was by debugging through the source code, trailing along and observing what was happening. Some stepping into's later, I stumbled upon this comment in line 469 in options.py pertaining to the node discover options in the options.py file:

[image]

It reads to me like this issue was known at some point, but I can't find anything about it in neither open nor closed issues.

Tried stepping much further in, but considering I'm not well versed in any of the source code, I was unable to pinpoint exactly what the issue was down the line. The bytearray command that was crafted in regards to the NO options seems okay to me, but has no effect. There were a few traceback errors in some fields a little later though:

[image]

Where the specific error messages were:
[image]
[image]

But I have no idea what that specfically means, nor if that's the issue or merely symptoms. Maybe the AT-command isn't being sent because of this? 
Either way, the errors didn't crash nor stop the modules from functioning.

tatianaleon, 2021-06-15:
Hi @Jawkawmaw, 

Could you provide an example to reproduce the problem?

This piece of code using an 802.15.4 XBee S2C as local XBee (firmware version 2003) discovers itself along with other nodes in the network if options enable `DiscoveryOptions.DISCOVER_MYSELF`, and do not discover itself when that option is not present.

For example, this piece of code:

```python
xbee = XBeeDevice(PORT, BAUD_RATE)

try:
    xbee.open()
    print("Local XBee: %s\n" % xbee)

    xnet = xbee.get_network()

    print("Original NO: %s" % utils.hex_to_string(xbee.get_parameter(ATStringCommand.NO)))
    xnet.set_discovery_options({DiscoveryOptions.DISCOVER_MYSELF})
    print("Modified NO: %s\n" % utils.hex_to_string(xbee.get_parameter(ATStringCommand.NO)))

    xnet.clear()

    def node_discovered_cb(remote):
        if remote == xbee:
            print("Myself discovered: %s" % remote)
        else:
            print("Device discovered: %s" % remote)

    xnet.add_device_discovered_callback(node_discovered_cb)
    xnet.start_discovery_process()

    print("Discovering remote XBee devices...")
    while xnet.is_discovery_running():
        time.sleep(0.1)

finally:
    if xbee is not None and xbee.is_open():
        xbee.close()

```

Its result is:

```
Local XBee: 0013A20041493F94 - usb5_2003

Original NO: 00
Modified NO: 01

Discovering remote XBee devices...
Myself discovered: 0013A20041493F94 - usb5_2003
Device discovered: 0013A2004195C70C - usb2_2002
Device discovered: 0013A2004195C887 - usb4_200C
```

If we change the line
```python
xnet.set_discovery_options({DiscoveryOptions.DISCOVER_MYSELF})
```
with
```python
xnet.set_discovery_options({})
```
to disable the "discover myself" option, the result is the expected one as well: only discovers the rest of nodes in the network, but not the local one:

```
Local XBee: 0013A20041493F94 - usb5_2003

Original NO: 01
Modified NO: 00

Discovering remote XBee devices...
Device discovered: 0013A2004195C70C - usb2_2002
Device discovered: 0013A2004195C887 - usb4_200C
```

Best Regards.

Jawkawmaw, 2021-06-21:
Well this is embarrassing; the callback works flawlessly now after **another** couple of power cycles, so that's no longer an issue. 
The original issue still persists, but I'm unsure if it's related to the AT command anymore. Anyways, I boiled it all down to as absolutely as simple as possible:

`local_device = XBeeDevice("COM4", 9600)
local_device.open()
xbee_network = local_device.get_network()
xbee_network.set_discovery_timeout(10)

xbee_network.set_discovery_options({DiscoveryOptions.DISCOVER_MYSELF})

xbee_network.start_discovery_process()
while xbee_network.is_discovery_running():
    sleep(0.1)

print(xbee_network.get_devices())`

Result is an empty list

tatianaleon, 2021-06-21:
Hi @Jawkawmaw,

Thank you for posting an example.

The method `get_devices()` of an `XBeeNetwork` only returns remote devices in the network. I mean, the network object never includes the local XBee, although you use the `DiscoveryOptions.DISCOVER_MYSELF` discovery option.

If you need to know when the local XBee answer is received, you must subscribe a device discovered callback for the network (`xbee_network.add_device_discovered_callback(node_discovered_cb)`), as it is done in the [example above](https://github.com/digidotcom/xbee-python/issues/245#issuecomment-861597893).

So, if your XBee is alone in the network, after a discovery process (with or without the `DiscoveryOptions.DISCOVER_MYSELF` option), your last statement `print(xbee_network.get_devices())` will result in an empty message.

Best Regards.

Jawkawmaw, 2021-06-21:
Hi Tatianaleon

Thanks for your quick response! So, just to make I understand this correctly, the network object "rejects" adding any discovered node if it is not remote?

Best regards

tatianaleon, 2021-06-22:
Hi @Jawkawmaw,

We could say that, the `XBeeNetwork` of an `XBeeDevice` only contains `RemoteXBeeDevices` (all of these applicable to subclasses) A local node will never be part of the node list in its network.

Of course, if you have more than one XBee connected to your machine, all of them part of the same network, and `XBeeDevice` instances for them, they will be remote nodes for the others and can be included in the others network as `RemoteXBeeDevices`.

That is, let's say you have 2 modules connected to your PC with node identifiers `xbee1` and `xbee2`, both in the same network, and a `XBeeDevice` instance for each: `local_xbee1` and `local_xbee2`. If you discovered nodes in `local_xbee1` network (`xnet1` ), a `RemoteXBeeNode` instance representing `xbee2` will be discovered and added to `xnet1` list of nodes, but not `local_xbee2` as `XBeeDevice` instance (and not `local_xbee1` nor a remote representation of `xbee1`, since it is the local node of the network)

Hope this helps.

Best Regards.

Jawkawmaw, 2021-06-22:
Ah, saying the network only contains RemoteXBeeDevices makes much more sense to me now, thanks for making me understand! 

Gonna close this issue now, have a great summer 😄

closed 2021-06-22 by Jawkawmaw