digidotcom/xbee-python #247: Using os.path.join() is not good for zipfile paths

DavePlater, 2021-07-06:
https://github.com/digidotcom/xbee-python/blob/2016bfb0bf27ad5469e5b2c9f231bd9719f88a2c/digi/xbee/profile.py#L893

(Trying to use the library on windows)
Using os.join() results in funky paths like `'radio_fw\\\\XB3-24A/XB3-24A_2003-th.xml'` which cannot be found and throw exceptions.
For looking at zipfile contents, i think you need to force the slash type

I changed to `self._fw_xml_file = _FW_DIR_NAME + "/" + self._fw_xml_filename` and the code executed fine

DavePlater, 2021-08-02:
I found this statement to back up my findings:

> Within a ZIP file, pathnames use the forward slash / as separator, as required by the ZIP spec (4.4.17.1).

[https://pkware.cachefly.net/webdocs/casestudies/APPNOTE.TXT](https://pkware.cachefly.net/webdocs/casestudies/APPNOTE.TXT)

> 4.4.17.1 The name of the file, with optional relative path.
       The path stored MUST NOT contain a drive or
       device letter, or a leading slash.  All slashes
       MUST be forward slashes '/' as opposed to
       backwards slashes '\' for compatibility with Amiga
       and UNIX file systems etc.  If input came from standard
       input, there is no file name field.  

So using os.path is the wrong thing for zip files

tatianaleon, 2021-08-13:
Thanks for reporting, @DavePlater. This issue is fixed in master (see commit https://github.com/digidotcom/xbee-python/commit/f2d9225965397c3a63530c5e5e2325c8d5325a72) and will be included in the next release.

In the meantime, you can download the source code from the master branch and install the library from there (see instructions at https://github.com/digidotcom/python-xbee/blob/master/README.rst#install-from-source).

Sorry for the inconvenience.

Best regards.

closed 2021-08-13 by tatianaleon