digidotcom/xbee-python #250: Could not determine operating mode.

hkayann, 2021-08-07:
Your example code `(SetAndGetParametersSample.py)` generating this issue while trying to run `device.open`. 

I was able to run my own code without any problems several months ago. But it seems like you changed the library in a way that requires setting operating mode beforehand, is it?

hkayann, 2021-08-07:
In the source code it mentions 
`InvalidOperatingModeException: If the XBee's operating mode is not
                API or ESCAPED API. This method only checks the cached value of
                the operating mode.`

So we can only create XBEEDevice if mode is set to API. How to create for transparent mode?

hkayann, 2021-08-07:
Apparently this library do not support transparent mode as it is mentioned. The workaround I've found is setting your Xbee to API mode via AT commands and then using this library.

closed 2021-08-07 by hkayann

tatianaleon, 2021-08-09:
Hi @hkayann,

The library only works with local XBee devices in API mode (with/without escapes).

You can pre-configure your local node to use an API mode, or use the `force_settings` option in the `open()` method.

With `force_settings=True`, the `open()` method establishes the serial connection with the specified serial settings, configuring the XBee with the selected serial values, and setting the operating mode to API without escapes (AP=1).

For example, to connect with the XBee attached to the port `/dev/ttyUSB0` using a baud rate of 115200, use:

```python
xbee = XBeeDevice(port="/dev/ttyUSB0", baud_rate=115200)
xbee.open(force_settings=True)
```

This configures the XBee to work in API without escapes (AP=1) and a baud rare of 115200 (BD=7) if it is not already, then opens the serial connection.

Best Regards.

hkayann, 2021-08-09:
Hi @tatianaleon,

Thanks for the explanation. This is what I am doing right now. I can communicate between several XBees connected to different Raspberry Pi's. However, the Arduino code, that I have written around 5 months ago is not working due to lack of Transparent mode support.

I was sending sensor data over serial (`Serial1.print(x)`) from Arduino Leonardo to Raspberry Pi. But now, I need a different library as XBees in API mode. There is one deprecated [library](https://github.com/andrewrapp/xbee-arduino) which is lastly updated 5 years ago.

Do you suggest anything regarding this, I am using Grove Bee Socket, XBee module, and Arduino Leonardo.

tatianaleon, 2021-08-09:
Hi @hkayann 

This library never supported Transparent communication with the local XBee, and the Arduino library you pointed out seems to only support API too.

It is difficult to suggest anything: I do not know what is your setup, what is the goal, how all your elements are connected, who sends and receives sensor data, etc.

Best Regards.

hkayann, 2021-08-10:
Hi @tatianaleon,

That means I was under the wrong impression when I used it 5 months ago.

Let me explain in more clear way.

The goal: Send sensor data from Arduino to Raspberry Pi over ZigBee.

This is the Arduino (transmitter) setup:
1. Grove Bee Socket
2. XBEE RF module
3. Grove Arduino shield.
4. Arduino Leonardo.

XBEE -> Grove Bee Socket -> Arduino Shield -> Arduino Leonardo

This is the Pi setup:
1. Raspberry Pi 4
2. XBEE RF Module

The Arduino code ( I am only putting the key part):

```
Serial1.print("ATCE 0\r"); // not a coordinator
Serial1.print("ATAP 0\r"); // run in Transparent mode
// set pan address, do everything needed than in loop
Serial1.print("send something");
```
The Python (receiver / coordinator) code, I am only putting related parts:
```
from digi.xbee.devices import XBeeDevice
from digi.xbee.exception import OperationNotSupportedException, XBeeException, RecoveryException
from digi.xbee.util import utils
from digi.xbee.models.status import NetworkDiscoveryStatus

PORT = "/dev/serial0"
BAUD_RATE = 9600
    device = XBeeDevice(PORT, BAUD_RATE)
    device.open(force_settings=True)
    while True:
        xbee_message = xbee_device.read_data()
        if xbee_message is not None:
            print("From %s >> %s" % (xbee_message.remote_device.get_64bit_addr(),
                                     xbee_message.data.decode()))
```

With this setup, Pi was receiving data from Arduino without any problem. So, how this setup worked, if the XBees were in different modes?

tatianaleon, 2021-08-10:
Hi @hkayann,

The setup you described should work now and also with previous versions of the library if the XBee connected to your Raspberry Pi is pre-configured to work in API mode (AP=1 for example) and without the `force_settings` argument that was not there.

The XBee modules that must work in API mode are those that are going to be used as local nodes (serially connected) by the XBee Python Library. In your setup:

- The XBee connected to the Raspberry Pi must be in API mode (because it is pre-configured or because you use the `force_settings=True` when opening the local node)
- The other XBee connected to the Arduino Leonardo can work in Transparent or API mode, this will depend on the code you are going to use. If you are transmitting raw data from/to the serial port where the XBee is connected, the operating mode of the node must be Transparent, as you have it.

Just a clarification, the operating mode (Transparent or API with/without escapes) "only" defines the way the XBee sends data over the serial port and the expected data format to read from the serial port. Data over the air is not affected by the selected operating mode. I mean XBee radios in different working modes can still communicate with each other over the air, but they are going to offer the received data over the serial connection using API frames (if in API mode) or in raw format (if in Transparent mode)

Best Regards.

hkayann, 2021-08-10:
Oh thank you @tatianaleon  very much for the clarification, it makes much more sense now. I got it wrong from the beginning.

Best Regards.