digidotcom/xbee-python #252: How to continue after error on serial port?

Artyrm, 2021-08-17:
If serial port being accessed by another application, or if it was briefly disconnected, device.reader gives an error, closes port and stops _packet_listener. But I want to reconnect after that. How to do it?

device.close() and device.open() sequence takes too much time (~1min) to use it.

tatianaleon, 2021-08-18:
Hi @Artyrm,

The serial port were an XBee is connected must only be used by one application. In this case, by the one using the library. There is no other mechanism to open/close the local connection with an XBee.

Best Regards.

Artyrm, 2021-08-18:
Hi @tatianaleon 

Thank you for your reply. 
Well, I believe since on any problems accessing serial reader (more precisely, `packet_listener`) is stopped (https://github.com/digidotcom/xbee-python/commit/02febb6d0be7d0f48db252f2a124ff1f120e69d7), there should be a way to restart it. It is critical, to say, on a device that run 24/7 unattended. 

For now I came up with that:
`   def run_reader(device):
      device._packet_listener.run()
      return True             
...
              if not device._packet_listener.is_running():
                    screen.print_at("Reader is down. See log for more info. Trying to rerun... ", 0, screen.height-1)
                    if not device.serial_port.is_open:
                        device.serial_port.open()
                    time.sleep(0.2)
                    if device.serial_port.is_open:
                        print("Serial port opened. ", end='')
                        screen.refresh()
                        task = threading.Thread(target=run_reader, args=[device], name="Run_reader", daemon=True)
                        screen.refresh()
                        task.start()
                        time.sleep(0.2)
                        if device._packet_listener.is_running():
                            print("Reader run again.", end='')
                            screen.refresh()`

Note the threaded run of `__packet_listener` that I'm doing from the other thread, because it is blocking. Maybe I'll add check for thread existence later, to prevent occasional threads multiplication. For now it behaves like OK solution, reconnects in less than a second.

Artyrm, 2021-08-19:
Well, https://github.com/digidotcom/xbee-python/issues/252#issuecomment-901416300 seems suitable.
If somebody will come up with better idea, please, feel free.

closed 2021-08-19 by Artyrm