digidotcom/xbee-python #256: ReadProfileException: Error reading profile: "There is no item named"

tuorpete, 2021-09-22:
I am trying to make my own remote device profile updater.
When I am loading the xpro-file (with for example "XBeeprofile(<path>)") I always get the "There is no item named for example 'radio_fw\\\\XB3-24Z/XB3-24Z_100A-th.xml' in the archive". I have checked that there is indeed file which is mentioned in the error.
I have tried to create multiple profiles with XCTU-NG and no matter what the device type is I am always getting basically the same error stating that the xml-file is missing.
I have checked that there is indeed file which is mentioned in the error.
What can I do?

tatianaleon, 2021-09-22:
Hi @tuorpete,

I guess you are using a Windows machine. There is a bug in version 1.4.0 solved in commit https://github.com/digidotcom/xbee-python/commit/f2d9225965397c3a63530c5e5e2325c8d5325a72 (see issue https://github.com/digidotcom/xbee-python/issues/247).

This fix will be included in the next release.

In the meantime, you can download the source code from the master branch and install the library from there (see instructions at https://github.com/digidotcom/python-xbee/blob/master/README.rst#install-from-source).

Sorry for the inconvenience.

Best Regards.

tuorpete, 2021-09-23:
Hi @tatianaleon 
Thank you for quick response.
Yes I use Windows. I installed version 1.4.1 and can see I now have the update in profile.py file. 
Doing for example: `profile = XBeeProfile('profile.xpro')` still gives the same error. I tried to put the profile in the same folder so I can use relative path and it does not fix it.

tatianaleon, 2021-09-23:
Hi @tuorpete

Could you provide the Python code that is failing and also the failure you are getting?
If you use the example https://github.com/digidotcom/xbee-python/tree/master/examples/profile/ReadXBeeProfileSample, setting the absolute path to your profile in `PROFILE_PATH`, what error do you get?

Best Regards.

tuorpete, 2021-09-23:
Thank you again @tatianaleon for quick response.

For ease of finding the problem I made a new venv, copied the example  from https://github.com/digidotcom/xbee-python/tree/master/examples/profile/ReadXBeeProfileSample as it is and only replaced the PROFILE_PATH with the filename that I have.
When in python console I issued command:
```
from main.py import main
main()
```
Then I get the following error:
`Error reading profile: "There is no item named 'radio_fw\\\\XB3-24Z/XB3-24Z_100A-th.xml' in the archive"`
The profile is generated with updated X-CTU NG and in this example to a XBee3 device.
I can see when opening the xpro-file with 7-ZIP I can see there is indeed a file called "XB3_24Z_100A-th.xml" in the path ".\test.xpro\radio_fw\XB3-24Z\"
Best Regards.

tatianaleon, 2021-09-23:
Thanks @tuorpete,

We will look for the issue. As soon as it is fixed, we will let you know.

Best Regards.

tuorpete, 2021-09-29:
Hi @tatianaleon 
Now the example program works. Great work :) Thank you very much.
I will now test the new library with my original program and report if there is any problems.

tuorpete, 2021-09-29:
Hi @tatianaleon,
I noticed that if I use just a file name (so the profile is in the same folder as the project) everything works fine. 
But when I use absolute path (e.g. 'C:\path\to\profile.xpro') I get:
 `ERROR: Error reading profile: encoding without a string argument
Traceback (most recent call last):
  File "<path to my project>\venv\lib\site-packages\digi\xbee\profile.py", line 943, in _initialize_profile
    self._parse_xml_firmware_file(zip_file)`
Do I need to first copy the profile to the same folder or is there an easy fix for this?

tatianaleon, 2021-09-29:
Hi @tuorpete,

No, you do not need to copy the profile.
I'm afraid I cannot reproduce the issue. Running the example https://github.com/digidotcom/xbee-python/tree/master/examples/profile/ReadXBeeProfileSample just changing the value of `PROFILE_PATH` to `C:/path/to/profile.xpro`, the code is working fine (using Python 3.9.7):
```
 +----------------------------------------------+
 | XBee Python Library Read XBee Profile Sample |
 +----------------------------------------------+

Profile information
-----------------------------------
  - Location: C:/path/to/profile.xpro
  - Version: 1
  - Firmware version: 4105
  - Hardware version: 66
  - Protocol: Zigbee
  - Flash firmware option: Flash firmware if it is different
  - Description: None
  - Reset settings: False
  - Has filesystem: False
  - AT settings:
    - Setting 'ID' - type: Number - format: No format - value: 1234
    - Setting 'JV' - type: Combo - format: No format - value: 1
    - Setting 'JN' - type: Combo - format: No format - value: 1
    - Setting 'NI' - type: Text - format: ASCII - value: my_xbee
    - Setting 'BD' - type: Combo - format: No format - value: 7
    - Setting 'AP' - type: Combo - format: No format - value: 1
    - Setting 'AO' - type: Number - format: No format - value: 1
```

tuorpete, 2021-09-30:
Hi @tatianaleon 
My bad. I found that I had basic problem with my path that I used only one '\' when I needed to have '\\' in the path's folder separators etc. 
Thank you for your help. I would say that this bug is now fixed.

closed 2021-09-30 by tuorpete