digidotcom/xbee-python #268: add_io_sample_received_callback() always using the same remote device

zgarnes17, 2022-02-15:
I am trying to create a network of multiple remote XBee 2 modules sending analog samples for a sensor network. I can discover two separate XBee devices with their corresponding MAC 64bit address IDs and the node IDs that I configured them to. I can receive the analog samples from the two different nodes, however, they keep saying the received signal is coming from the same node. I know this isn't true seeing I know there are two different analog signals on each node. Code and the output are below. Please help! 

CODE: 
`import serial
import time
from digi.xbee.devices import XBeeDevice
from digi.xbee.io import IOLine, IOMode
import mysql.connector
import datetime
import csv
import MySQLdb
# Serial port on Raspberry Pi
SERIAL_PORT = "/dev/ttyS0"

# BAUD rate for the XBee module connected to the Raspberry Pi
BAUD_RATE = 9600

# Analog pin we want to monitor/request data
ANALOG_LINE1 = IOLine.DIO1_AD1

# Sampling rate
SAMPLING_RATE = 15

# Get an instance of the XBee device class 
device = XBeeDevice(SERIAL_PORT, BAUD_RATE)

# Method to connect to the network and discover the nodes
def discover_nodes():
    """Get a list of the nodes (node ids) on the network
    Returns:
    """
    # Request the network class and search the network for the remote node
    xbee_network = device.get_network()
    xbee_network.start_discovery_process()
    print("Discovering network", end='')
    while xbee_network.is_discovery_running():
        print(".", end='')
        time.sleep(0.5)
    print("done.")
    devices = xbee_network.get_devices();
    node_ids= []
    for dev in devices:
        print("Found {0} at {1}.".format(dev.get_node_id(), dev.get_64bit_addr()))
        node_ids.append(dev.get_node_id())
    if not node_ids:
        print("WARNING: No nodes found.")
    return node_ids

# Method to connect to the network and get the remote node by id
def get_remote_devices(remote_node_ids):
    """Get the remote node from the network
    Returns:
    """
    # Request the network class and search the network for the remote node
    xbee_network = device.get_network()
    remote_device = xbee_network.discover_devices(remote_node_ids)
    if remote_device is None:
        print("ERROR: Remote node id {0} not found.".format(remote_id))
        exit(1)

# Method to get the data when available from the remote node
def io_sample_callback(sample, remote, time):
    address = str(remote.get_64bit_addr())
    # Get the raw temperature value
    raw_temp = sample.get_analog_value(ANALOG_LINE1)
    # Save results in the table
    short_addr = address[-4:]
    # Get the temperature in Celsius
    temp_c = (raw_temp / 1023.0 * 1.25 - 0.5) * 100.0
    # Calculate temperature in Fahrenheit
    temp_f = ((temp_c * 9.0) / 5.0) + 32.0
    print("\tTemperature is {0:.2f}C. {1:.2f}F from node {2}".format(temp_c, temp_f,remote.get_64bit_addr()))
try:
    # Read and save temperature data
    print("Welcome to example of storing data from a set of remote TMP36 sensors in MySQL!")
    device.open()  # Open the device class
    # Get the nodes on the network
    remote_node_ids = discover_nodes()
    
    # Setup the remote device
    get_remote_devices(remote_node_ids)
    device.add_io_sample_received_callback(io_sample_callback)
    # Register a listener to handle the samples received by the local device         
    while True:
        pass
except KeyboardInterrupt:
    if device is not None and device.is_open():
        device.close()
# Disconnect from the server
try:
    db_conn.disconnect()
except:
    pass
`

OUTPUT: 
Welcome to example of storing data from a set of remote TMP36 sensors in MySQL!
Discovering network......done.
Found South Node at 0013A20041E57931.
Found North Node at 0013A20041E57938.
	Temperature is 21.97C. 71.55F from node 0013A20041E57931
	Temperature is -50.00C. -58.00F from node 0013A20041E57931
	Temperature is 21.97C. 71.55F from node 0013A20041E57931
	Temperature is -50.00C. -58.00F from node 0013A20041E57931

tatianaleon, 2022-02-16:
Hi @zgarnes17,

Are your XBee modules 802.15.4? If so, did you configure different 16-bit addresses for your modules (`MY` parameter)? Or all of them to `0xFFFE`?
In 802.15.4, to use long 64-bit address, `MY` must be `0xFFFE` or `0xFFFF`, otherwise modules use short source addressing (`MY` address).
By default, `MY` is `0x0000` in 802.15.4 firmware, so, by default, modules use short 16-bit addressing and `0x0000` as address. 

If this is the case, both discovered nodes `0013A20041E57931` and `0013A20041E57938` have the same 16-bit address, the default one, `0x0000`. So when an IO message is received from any of them, the source address is `0x0000` always and the library looks for the first discovered node with that 16-bit address.

in 802.15.4, for 16-bit addressing, you must configure all the nodes in the network with different `MY` values. If you prefer to use 64-bit addresses, then set the `MY` of all your modules to `0xFFFE` or `0xFFFF`.

See https://www.digi.com/resources/documentation/Digidocs/90001500/Default.htm#Reference/r_addressing_802.htm

Best Regards

zgarnes17, 2022-02-16:
Hello @tatianaleon 

Thank you much!!! That worked and solve so many headaches for me. Thanks for the fast response and the help!

With gratitude, Zgarnes

tatianaleon, 2022-02-17:
You are welcome :-)

closed 2022-02-17 by tatianaleon