digidotcom/xbee-python #271: Accessing LQI value

Clif123, 2022-03-14:
Hi all,

Using deep network discovery and then printing out the connection using get_connections() gives me this (an example),

Code:
print("%s" % '\n'.join(map(str, xnet.get_connections())))

Output:
{0013A20041B4954A - X3 >>> 0013A20041C7DAC2 - X1 [Active / Active]: 224/226}

How can I extract the lqi_a2b and lqi_b2a individually? Everytime I try to print the get_connections, what I would get is the address location of the object. 

Thanks in advance.

tatianaleon, 2022-03-15:
Hi @Clif123,

You can iterate the connections in the list and use the `lq_a2b` and `lq_b2a` properties of each one. They are `LinkQuality` objects and have `lq` (integer) and `is_rssi` (boolean) as properties:

```python
for c in xnet.get_connections():
    print("%s >>> %s: %d / %d\n" % (c.node_a, c.node_b, c.lq_a2b.lq, c.lq_b2a.lq))
```

Best Regards

Clif123, 2022-03-15:
> ```python
> for c in xnet.get_connections():
>     print("%s >>> %s: %d / %d\n" % (c.node_a, c.node_b, c.lq_a2b.lq, c.lq_b2a.lq))
> ```

This works. Thanks so much!

closed 2022-03-15 by Clif123