digidotcom/xbee-python #277: How can I change AP mode to 0 or 2?

KyleIm, 2022-08-02:
I am using XBee PRO S3B for wireless radio communication.

I am curious that is there any way I can open or change my AP to 0 or 2.

To avoid confusion here is my source code. This is for resetting the destination mac address.

import digi
from digi.xbee.devices import XBeeDevice
xbee0=XBeeDevice("/dev/ttyUSB0",9600)
xbee0.open(force_settings=True)
xbee0.reset()

The problem of this code would force me to use AP = 1 only. Is there any way I can change it back to AP = 0 or AP = 2 after I reset everything? (Currently I have no idea other than using XCTU. But I want to do it using this library.)

tatianaleon, 2022-08-08:
Hi @KyleIm,

To change the operating mode of your XBee you can use XCTU for example. Using the library you can only use API modes, that is `AP=1` or `AP=2`, since transparent mode (`AP=0`) is not supported. You can use:
```python
from digi.xbee.models.atcomm import ATStringCommand
from digi.xbee.models.mode import OperatingMode

[...]

# Set operating mode to 1 (API without escapes)
xbee0.set_parameter(ATStringCommand.AP, bytearray([OperatingMode.API_MODE.code]))

# Set operating mode to 2 (API with escapes)
xbee0.set_parameter(ATStringCommand.AP, bytearray([OperatingMode.ESCAPED_API_MODE.code]))

[...]
```

The method `open(force_settings=True)` of an `XBeeDevice` object forces the XBee configuration to the one provided when you instances it. In your case to 9600 bauds, 8 data bits, no parity, 1 stop bit, and no flow control. It also forces the operating mode of the XBee to `AP=1` if it is not configured to use an API mode (that is, if `AP!=1` and `AP!=2`)
See [Open the XBee connection](https://xbplib.readthedocs.io/en/stable/user_doc/working_with_xbee_classes.html#open-the-xbee-connection).

The `reset()` method does not restore the configuration of your XBee. It performs a software reset of the module, it executes the command `FR`. See [Reset the device](https://xbplib.readthedocs.io/en/stable/user_doc/configuring_the_xbee_device.html?highlight=reset#reset-the-device).

To restore default values for your module you must execute `RE` command (see [RE (Restore Defaults)](https://www.digi.com/resources/documentation/digidocs/90001539/#reference/r_cmd_re.htm)). After restoring:

1. To permanently store changes on the XBee you must write them (`write_changes()`)
2. Reset the module so it use the established default configuration (`reset()`)
3. Then close the connection (`close()`)

If after this, you need to continue communicating with the XBee, you must open again the connection. This means, after using `open(force_settings=True)`, some parameters will be modified to use the communication parameters you set on the object instance and the `AP` setting value to be 1, since the default value, 0 (transparent mode) is not supported by the library (see [Open the XBee connection](https://xbplib.readthedocs.io/en/stable/user_doc/working_with_xbee_classes.html#open-the-xbee-connection))

For example:

```python
import sys

from digi.xbee.models.atcomm import ATStringCommand
from digi.xbee.devices import XBeeDevice
from digi.xbee.exception import XBeeException

DEFAULT_NODE_ID = " "
NODE_ID = "xbee0"

def restore_defaults_and_close(xbee):
    try:
        xbee.execute_command(ATStringCommand.RE, apply=False)
        xbee.write_changes()
        print("XBee default configuration restored!")
    except XBeeException as exc:
        print("Error restoring XBee default configuration: %s" % str(exc))
        sys.exit(1)

    try:
        xbee.reset()
    except XBeeException as exc:
        # This is expected if the previous XBee connection parameters
        # are different to the default ones
        print("   [Expected] Error resetting the XBee: %s" % str(exc))
    finally:
        xbee.close()

def open_xbee(xbee):
    try:
        xbee.open(force_settings=True)
    except XBeeException as exc:
        print("Unable to establish communication with XBee: %s" % str(exc))
        sys.exit(1)

def run():
    xbee0 = XBeeDevice("/dev/ttyUSB5", 115200)
    open_xbee(xbee0)

    # Set node ID to a non default value, i.e., xbee0
    try:
        xbee0.set_node_id(NODE_ID)
        xbee0.write_changes()
        xbee0.apply_changes()
        # Read the node ID to test the value
        node_id = xbee0.get_node_id()
        print("\nNode id BEFORE restoring settings: '%s'\n" % node_id)
        if node_id != NODE_ID:
            print("ERROR: Expected value '%s', retrieved '%s'" % (NODE_ID, node_id))
            sys.exit(1)
    except XBeeException as exc:
        print("Unable to set XBee node ID: %s" % str(exc))
        sys.exit(1)

    # Restore defaults and close the connection
    restore_defaults_and_close(xbee0)

    # Here you can reopen the device, but some configuration parameters will
    # be changed. For example:
    #    * the baudrate to be 115200 (since 'xbee0' was instantiated to use
    #      115200 instead the defaults 9600)
    #    * or the operating mode (AP), since the default mode, transparent
    #      (AP=0) is not supported by the library. The XBee will be reconfigured
    #      to use API without escapes (AP=1).
    print("Re-opening XBee")
    open_xbee(xbee0)

    # Read the node ID, it should be a white space (the default value)
    node_id = xbee0.get_node_id()
    print("\nNode id AFTER restoring settings: '%s'\n" % node_id)
    if node_id != DEFAULT_NODE_ID:
        print("ERROR: Expected value '%s', retrieved '%s'" % (DEFAULT_NODE_ID, node_id))
        sys.exit(1)

    sys.exit(0)

if __name__ == '__main__':
    run()
```

The output is:
```
Node id BEFORE restoring settings: 'xbee0'

XBee default configuration restored!
   [Expected] Error resetting the XBee: Timeout waiting for the modem status packet.
Re-opening XBee

Node id AFTER restoring settings: ' '
```

Hope this helps.

KyleIm, 2022-08-08:
Thanks I solved it. However, **xbee0.set_parameter(ATStringCommand.AP, bytearray([OperatingMode.API_MODE.code]))** should be **xbee0.set_parameter(('AP'), bytearray([OperatingMode.API_MODE.code]))**

When I copied it, it showed **TypeError: object of type 'ATStringCommand' has no len()**.

closed 2022-08-08 by KyleIm

tatianaleon, 2022-08-09:
Great @KyleIm!

Using `ATStringCommand.AP` in `get_parameter()`,  `set_parameter()`, and `execute_parameter()` methods should work since version 1.4.0 of the library.

Best Regards.